modelica/ModelicaStandardLibrary #4075: Icon layout of TerminateSimulation similar to 'expression' sources

tobolar, 2023-02-27:
The layout of icon of `Blocks.Logical.TerminateSimulation` shall be defined similar to other 'expression' sources (e.g. `Blocks.Sources.RealExpression`). This esp. concerns icon size and name text. The icon code can be like:
```
    Icon(
      coordinateSystem(
        preserveAspectRatio=true,
        extent={{-200,-100},{200,100}}),
      graphics={
        Rectangle(fillColor={235,235,235},
          fillPattern=FillPattern.Solid,
          lineThickness=5,
          borderPattern=BorderPattern.Raised,
          extent={{-200,-40},{200,40}}),
        Text(extent={{-166,-15},{194,15}},
          textString="%condition"),
        Rectangle(fillColor={161,35,41},
          fillPattern=FillPattern.Solid,
          borderPattern=BorderPattern.Raised,
          extent={{-194,-14},{-168,14}}),
        Text(textColor={0,0,255},
          extent={{-200,90},{200,50}},
          textString="%name")})
```

This change would significanlty influence diagram layouts of classes where `TerminateSimulation` block is used. Therefore, a *conversion script* shall be provided to resize the element instance and, thus, to retain original diagram layout.

References #3986

HansOlsson, 2023-02-27:
I don't fully understand this:
- There is currently no support for resizing elements in conversion scripts.
- The goal of #3986 was to no longer preserve the aspect ratio of this block - the code above has it.
- The original diagram layout for models in MSL using TerminateSolution do not respect preserveAspectRatio. What should we retain?

tobolar, 2023-02-27:
In the suggested code, there is
1. Icon size 400x200 (i.e. `{{-200,-100},{200,100}}`), originally 400x40 and
2. no `initialScale` anymore.
 
So an instance in existing models/classes would look differently.

>  to no longer preserve the aspect ratio of this block - the code above has it.

Sorry, my mistake - it of course shall *not* preserve the aspect ratio. Fixed this in the code above.