modelica/ModelicaStandardLibrary #4076: Modelica.StateGraph.Examples.Utilities.Tank could set units for its variables.

qlambert-pro, 2023-03-03:
In particular:

```
  Real level(start = 0, fixed = true, unit = "1") "Tank level in % of max height";
  parameter Real A(unit = "m2") = 1 "Ground area of tank in m^2";
  parameter Real a(unit = "m2") = 0.2 "Area of drain hole in m^2";
  parameter Real hmax(unit = "m") = 1 "Max height of tank in m";
  ```

beutlich, 2024-01-14:
I would not add the unit in the desciption comment `... in m^2` since the user can change the display unit as desired.

henrikt-ma, 2024-01-15:
> I would not add the unit in the description comment `... in m^2` since the user can change the display unit as desired.

Right.  Not because the statement would technically be wrong if interpreted as applying to the `unit`, but both because of the confusion it may causes in the presence of `displayUnit` and because of the not-so-good-looking redundancy when the description is displayed in a contexts where the parameter value is already displayed with the unit.

henrikt-ma, 2024-01-15:
> I would not add the unit in the description comment `... in m^2` since the user can change the display unit as desired.

This is a pattern that should be fixed more generally in the entire library.  The regexp ` in [a-zA-Z0-9^/.]*"` gives lots of examples (but also meny matches which have nothing to do with redundant units).

PR #4264 (2024-01-18): Set unit instead of having it in description
Closes #4076 
Only that StateGraph model and two uses of "in dB" were changed.
In addition to these changes the only other cases were in Modelica.Units.Conversions - cases such as the following:

```
function to_degC "Convert from kelvin to degree Celsius"
  extends Modelica.Units.Icons.Conversion;
  input SI.Temperature Kelvin "Value in kelvin";
  output Modelica.Units.NonSI.Temperature_degC Celsius "Value in degree Celsius";
```

I **didn't** update them, because:
- It seemed a bit tedious
- Without the unit the documentation would be kind of mean [... cut]
commit 24477ff6 (2024-01-18): Set unit instead of having it in description (#4264)

* Replace description with unit-setting.
Ideally it should use SI.Area etc, but I thought the idea was that the example should be more self-contained.
Closes #4076

* Also remove "in dB" since the unit is "dB".

* Update Modelica/StateGraph.mo

* Update Modelica/StateGraph.mo

---------
closed 2024-01-18 by Harisankar-Allimangalath