modelica/ModelicaStandardLibrary #4079: Unit error in CauerLowPassAnalog

qlambert-pro, 2023-03-03:
Both values of capacitance c2 and c4 are expressed as inverse of inductance l1 and l2 respectively.

qlambert-pro, 2023-03-28:
`Modelica.Electrical.Analog.Examples.CauerLowPassSC`  suffers from the exact same issue.

HansOlsson, 2023-03-30:
That makes it sound worse than it is - as they are expressed as:
`parameter SI.Capacitance c2=1/(1.704992^2*l1);`
so it would make sense if `1.704992` had unit `Hz` (and similarly for `c4`).

However, that made me realize that there's a lot of documentation missing for these examples - it's stated that it's a 5th order low-pass filter (obviously a Cauer or elliptic filter), but it doesn't specify the coefficients like frequency, ripple, and selectivity.

christiankral, 2023-06-01:
Some possible solutions are:
1. Remove the example as the documentation is not sufficient at all
2. Just replace `parameter SI.Capacitance c2=1/(1.704992^2*l1)` by `parameter SI.Capacitance c2=1/(1.704992^2*1.304)` and leave the example as it is, as it is "just" an example
3. Replace the example by a new and well documented example of such a fifth order filter: this sounds like a lot of work (for an example that I personally think is not too exciting at all ...)