modelica/ModelicaStandardLibrary #4080: Unit error in Modelica.Mechanics.MultiBody.Examples.Loops.Utilities.GasForce2

qlambert-pro, 2023-03-03:
For `R_air`:
```
  constant SI.SpecificHeatCapacity R_air = Modelica.Constants.R / 0.0289651159;
```
we should introduce the molar mass of air, if I am not mistaken.

tobolar, 2023-03-03:
Do you mean **SI units**?

Considering `dens = press/(R_air*T);`, the SI unit of `R_air` (J/(kg.K)) seems to be fine.

tobolar, 2023-03-03:
Something like(?)
```
  constant SI.MolarMass MM_air = 0.0289651159;
  constant SI.SpecificHeatCapacity R_air = Modelica.Constants.R/MM_air;
```

qlambert-pro, 2023-03-03:
That's what I had in mind. If we go by this proposal: https://github.com/modelica/ModelicaSpecification/pull/3266#discussion_r1019240597, this is the sort of things tools should start rejecting.

PR #4081 (2023-05-17): Introduce molar mass for better balance of SI units
Close #4080
closed 2023-05-17 by MartinOtter