modelica/ModelicaStandardLibrary #4086: Some types in Units don't specify quantity

henrikt-ma, 2023-03-09:
Is it intentional that a type like this one doesn't come with a `quantity`?
https://github.com/modelica/ModelicaStandardLibrary/blob/ef08d88cfddd2bd0104b987b6db70327b359abde/Modelica/Units.mo#L346

beutlich, 2024-01-16:
Looks to me that only `Stress` is affected (and some ISO-chapter 4 types).

> Is it intentional that a type like this one doesn't come with a `quantity`?

Yes, according to https://github.com/modelica/ModelicaStandardLibrary/blob/ef08d88cfddd2bd0104b987b6db70327b359abde/Modelica/Units.mo#L1899

Not sure if this reasoning still holds today with more than one Modelica tool supporting Units.

henrikt-ma, 2024-01-16:
From the cited line:
> Quantity="Stress" removed from type "Stress", in order that a type "Stress" can be connected to a type "Pressure".

> Not sure if this reasoning still holds today with more than one Modelica tool supporting Units.

It looks like a completely valid reason, independently of tool.  If one really wants to be able to have a `connect`-equation with a `Stress` on one side and a `Pressure` on the other side, the two must not have different `quantity`.  This is the _only_ semantical consequence of specifying `quantity` in Modelica.

henrikt-ma, 2024-01-16:
By the way, whatever the explanation is, I think it belongs in a comment at the definition of `Stress` rather than in (or as a complement to) the class documentation.