modelica/ModelicaStandardLibrary #4088: Are "CircularWaveNumber" and "CircularWavenumber" really two different quantities?

henrikt-ma, 2023-03-09:
It seems very confusing to offer both of these:
```
    type CircularWavenumber = Real (final quantity="CircularWavenumber", final unit="m-1");
    type CircularWaveNumber = Real (final quantity="CircularWaveNumber", final unit="rad/m");
```

If it is intentional to have two different `quantity` strings for these, I suggest to at least clearly comment in the code why we have both.

dietmarw, 2023-03-09:
Looks like a pure oversight, most likely due to the non-alphabetic sorting of types which makes it harder to see these. Question is, which of the units should be kept and which should be a pointer (needs to be at least until the next MAJOR version).

beutlich, 2023-08-08:
@AHaumer introduced both unit types by 2e76e29aaa2bf6e2f8c30a342b5a72fdff7c90e7.

HansOlsson, 2023-08-08:
> @AHaumer introduced both unit types by [2e76e29](https://github.com/modelica/ModelicaStandardLibrary/commit/2e76e29aaa2bf6e2f8c30a342b5a72fdff7c90e7).

No, that just touched them.

Before that Dag Brück touched both in 6a90d1a893 and before that in [7366c2](https://github.com/modelica/ModelicaStandardLibrary/commit/7366c28985d642838fadbe8fae46305e9c62b98f) under the name SIunits.mo. That is just version 1.3.1 from 1999 - and we don't have git history before that.

DagBruck, 2023-08-18:
I took the liberty of unassigning myself.