modelica/ModelicaStandardLibrary #4093: Provide CMake configuration as proof of concept for future standardization

henrikt-ma, 2023-03-14:
This issue is opened on behalf of the MAP-Lang design meeting taking place this week.

On the topic of improving the handling of external code in Modelica, the language group wants to go in the direction of standardizing on the use of CMake for building external code:
- https://github.com/modelica/ModelicaSpecification/issues/1668#issuecomment-1468428017

We believe that instead of starting with standardizing this in the Modelica specification, it would be better to first work out a good CMake configuration for the MSL, that tool vendors can try out and contribute to if they want to.  When the MAP-Lib and interested tool vendors are satisfied with the setup, we can then proceed with standardization in the language group.

The idea isn't new, but it seems it hasn't been discussed for the 10 last years:
- https://github.com/modelica/ModelicaStandardLibrary/issues/1155
- https://github.com/modelica/ModelicaStandardLibrary/issues/1067#issuecomment-272630259

Maybe someone could shed some light on what CMake file that this comment from 2016 refers to?
- https://github.com/modelica/ModelicaStandardLibrary/issues/1856#issuecomment-272677300

henrikt-ma, 2023-03-14:
@maltelenz, would you be interested in providing something based on what we do at Wolfram?

beutlich, 2024-01-14:
> Maybe someone could shed some light on what CMake file that this comment from 2016 refers to?

I guess it should read Makefile (instead of CMake-file).

henrikt-ma, 2024-01-15:
> > Maybe someone could shed some light on what CMake file that this comment from 2016 refers to?
> 
> I guess it should read Makefile (instead of CMake-file).

Thanks.  Let's consider this item resolved.

beutlich, 2024-01-15:
There now is #4265.

PR #4265 (2024-08-01): Add CMake configuration for external C-Sources
Resolves #4093.

This can be used as base for a discussion on CMake configuration.

TODO:
- [x] Darwin specifics
- [x] Compiler specific library subdirectory
- [x] Test in CI (GitHub Actions)
- [x] Unix CFLAGS?
- [x] CTest
- [x] Location of top-level CMakeLists.txt: \<LibName\>/Resources/CMakeLists.txt
- [x] Minimum CMake version: 3.15

Note, that this always builds with the _static_ MSVC runtime library (required by Dymola on Win) which best is handled by CMake&nbsp;&GreaterEqual;&nbsp;3.15.
closed 2024-08-01 by casella