modelica/ModelicaStandardLibrary #4102: Invalid record type passed to functions in Fittings

bilderbuchi, 2023-03-29:
> **Comment by sjoelund.se on 4 Apr 2014 08:02 UTC** I think this would be rather important to resolve or clarify. Currently the MSL does not seem to care about the order: `CurvedBend.Geometry` and `Bend.dp_curvedOverall_IN_con` need to be type-compatible to translate some Fluid.Dissipation models, and the records are:
> 
> ```
> record EdgedBend "Input for bend"
>   SI.Diameter d_hyd;
>   SI.Angle delta;
>   SI.Length K;
> end EdgedBend;
> 
> record Bend "Input for bend"
>   extends EdgedBend;
>   SI.Radius R_0;
> end Bend;
> 
> record dp_curvedOverall_IN_con
>   extends Modelica.Fluid.Dissipation.Utilities.Records.PressureLoss.Bend;
>   // d_hyd,delta,K,R_0
> end dp_curvedOverall_IN_con;
> 
> record Geometry "Geometric data for a curved bend"
>   SI.Diameter d_hyd;
>   SI.Radius R_0;
>   SI.Angle delta;
>   Modelica.Fluid.Types.Roughness K;
> end Geometry;
> ```
> 
> There are also some user libraries out there that do precisely the `r0 := r` assignment.

This issue with `CurvedBend` has apparently never been fixed.
I'm getting an "error: passing 'Modelica_Fluid_Fittings_BaseClasses_Bends_CurvedBend_Geometry' to parameter of incompatible type 'Modelica_Fluid_Dissipation_PressureLoss_Bend_dp__curvedOverall__IN__con'" on compilation in OpenModelica, and from what I can tell the situation sketched above by @sjoelund is still present in MSL 4 (so I don't think it's an OpenModelica problem at this point).

_Originally posted by @bilderbuchi in https://github.com/modelica/ModelicaStandardLibrary/issues/1245#issuecomment-1486446524_

bilderbuchi, 2023-03-29:
@beutlich advised to open this issue, and has identified a list of locations to modify:
> > Is it correct that this issue with `CurvedBend` has never been fixed?
> 
> I guess you are right here.
> 
> The invalid assignments are:
> 
> https://github.com/modelica/ModelicaStandardLibrary/blob/64b1f5716d8bf630ce0aea0263248bf89d2d32df/Modelica/Fluid/Fittings.mo#L17
> 
> https://github.com/modelica/ModelicaStandardLibrary/blob/64b1f5716d8bf630ce0aea0263248bf89d2d32df/Modelica/Fluid/Fittings.mo#L30
> 
> https://github.com/modelica/ModelicaStandardLibrary/blob/64b1f5716d8bf630ce0aea0263248bf89d2d32df/Modelica/Fluid/Fittings.mo#L2118
> 
> https://github.com/modelica/ModelicaStandardLibrary/blob/64b1f5716d8bf630ce0aea0263248bf89d2d32df/Modelica/Fluid/Fittings.mo#L2167
> 
> @bilderbuchi Could you please open a new issue, referencing this old issue, such that we do not need to reopen this issue. Thanks.

TManikantan, 2023-06-27:
After reviewing comments #1245 and [#10803](https://github.com/OpenModelica/OpenModelica/issues/10803),it is evident that the resolution of [#10803](https://github.com/OpenModelica/OpenModelica/issues/10803) will naturally address the present problem. @bilderbuchi @casella do you suggest to wait till [#10803](https://github.com/OpenModelica/OpenModelica/issues/10803) materializes.

beutlich, 2023-06-27:
I am afraid I do not understand your comment as this is a library issue and not a tool issue.

casella, 2023-06-28:
@beutlich as I understand the record type is indeed valid according to the MLS, because it contains the required variables, but it contains them in a different order, because of the way the type was defined. OMC currently cannot handle that, but it's a tool issue, not a library issue.

casella, 2025-02-11:
Tool issue.

closed 2025-02-11 by casella