modelica/ModelicaStandardLibrary #4132: Link check returns error 403

tobolar, 2023-05-16:
There is error 403 during link checking regarding the URL link https://www.tandfonline.com/doi/full/10.1080/13873954.2020.1757726 in ThermofluidStream, see action https://github.com/DLR-SR/ThermofluidStream/actions/runs/4981391541/jobs/8940252272.
Even if this is not the link used in MSL, the CI/Python functions used are the same as for MSL. So I wonder whether there is a solution available to handle the error 403 properly.
For first, I tried to enhance the User-Agent to `Mozilla/5.0 (Windows NT 6.1; Win64; x64; rv:47.0) Gecko/20100101 Firefox/47.0` but this was not successful.

So maybe some post-process handling would better suit here.

beutlich, 2023-05-24:
Hm, cannot confirm your observation with the User-Agent.

``` bash
curl https://www.tandfonline.com/doi/full/10.1080/13873954.2020.1757726 -i
```

returns 403, but

``` bash
curl -A "Mozilla/5.0 (Windows NT 6.1; Win64; x64; rv:47.0) Gecko/20100101 Firefox/47.0" https://www.tandfonline.com/doi/full/10.1080/13873954.2020.1757726 -i
```

returns 302. You might also use the proper DOI and get around the User-Agent customization:

``` bash
curl https://doi.org/10.1080/13873954.2020.1757726 -i
```

return 302.

tobolar, 2023-06-12:
> You might also use the proper DOI URL and get around the User-Agent customization

@beutlich  Thanks for this hint - I have changed this now.

I will close this issue, even if the potential problem of the error 403 remains unresoved to me.

closed 2023-06-12 by tobolar