modelica/ModelicaStandardLibrary #4138: Proposal - Add new VariableTranslation component to MSL

BasilioPV, 2023-05-23:
I have created a simple Variable Translation component based on a translation with a vector[3] real input.
[image]

This component is very useful to place forces over a body that their application point changes over time. The 3D animation was disabled.

If this has interest to add to MSL, please indicate on the issue how can I help on this. Oterwise close the issue.
Thanks!

Source code:

model VariableTranslation "Derivative work of Translation component of MSL"
  import Modelica.Mechanics.MultiBody.Types;
  import Modelica.SIunits.Conversions.to_unit1;
  Modelica.Mechanics.MultiBody.Interfaces.Frame_a frame_a "Coordinate system fixed to the component with one cut-force and cut-torque" annotation(
    Placement(transformation(extent = {{-116, -16}, {-84, 16}})));
  Modelica.Mechanics.MultiBody.Interfaces.Frame_b frame_b "Coordinate system fixed to the component with one cut-force and cut-torque" annotation(
    Placement(transformation(extent = {{84, -16}, {116, 16}})));
  Modelica.Blocks.Interfaces.RealInput r_vect[3] annotation(
    Placement(visible = true, transformation(origin = {-4, -106}, extent = {{-20, -20}, {20, 20}}, rotation = 90), iconTransformation(origin = {-2, -100}, extent = {{-20, -20}, {20, 20}}, rotation = 90)));
protected
  outer Modelica.Mechanics.MultiBody.World world;
equation
  Connections.branch(frame_a.R, frame_b.R);
  assert(cardinality(frame_a) > 0 or cardinality(frame_b) > 0, "Neither connector frame_a nor frame_b of FixedTranslation object is connected");
  frame_b.r_0 = frame_a.r_0 + Modelica.Mechanics.MultiBody.Frames.resolve1(frame_a.R, r_vect);
  frame_b.R = frame_a.R;
// Force and torque balance
  zeros(3) = frame_a.f + frame_b.f;
  zeros(3) = frame_a.t + frame_b.t + cross(r_vect, frame_b.f);
  annotation(
    Icon(coordinateSystem(preserveAspectRatio = true, extent = {{-100, -100}, {100, 100}}), graphics = {Text(textColor = {0, 0, 255}, extent = {{-150, 85}, {150, 45}}, textString = "%name"), Text(extent = {{150, -50}, {-150, -20}}, textString = "r=%r_vect"), Text(textColor = {128, 128, 128}, extent = {{-89, 38}, {-53, 13}}, textString = "a"), Text(textColor = {128, 128, 128}, extent = {{57, 39}, {93, 14}}, textString = "b"), Line(origin = {-10, 0}, points = {{10, 20}, {-10, -20}}), Line(origin = {10, 0}, points = {{10, 20}, {-10, -20}}), Polygon(origin = {-54, 0}, fillPattern = FillPattern.Solid, points = {{-46, 4}, {46, 4}, {42, -4}, {-46, -4}, {-46, 4}}), Polygon(origin = {54, 0}, rotation = 180, fillPattern = FillPattern.Solid, points = {{-46, 4}, {46, 4}, {42, -4}, {-46, -4}, {-46, 4}})}),
    Diagram(coordinateSystem(preserveAspectRatio = true, extent = {{-100, -100}, {100, 100}}), graphics = {Rectangle(fillPattern = FillPattern.Solid, extent = {{-100, 5}, {100, -5}}), Line(points = {{-95, 20}, {-58, 20}}, color = {128, 128, 128}, arrow = {Arrow.None, Arrow.Filled}), Line(points = {{-94, 18}, {-94, 50}}, color = {128, 128, 128}, arrow = {Arrow.None, Arrow.Filled}), Text(textColor = {128, 128, 128}, extent = {{-72, 35}, {-58, 24}}, textString = "x"), Text(textColor = {128, 128, 128}, extent = {{-113, 57}, {-98, 45}}, textString = "y"), Line(points = {{-100, -4}, {-100, -69}}, color = {128, 128, 128}), Line(points = {{-100, -63}, {90, -63}}, color = {128, 128, 128}), Text(textColor = {128, 128, 128}, extent = {{-22, -39}, {16, -63}}, textString = "input r"), Polygon(fillPattern = FillPattern.Solid, points = {{88, -59}, {88, -68}, {100, -63}, {88, -59}}), Line(points = {{100, -3}, {100, -68}}, color = {128, 128, 128}), Line(points = {{69, 20}, {106, 20}}, color = {128, 128, 128}, arrow = {Arrow.None, Arrow.Filled}), Line(points = {{70, 18}, {70, 50}}, color = {128, 128, 128}, arrow = {Arrow.None, Arrow.Filled}), Text(textColor = {128, 128, 128}, extent = {{92, 35}, {106, 24}}, textString = "x"), Text(textColor = {128, 128, 128}, extent = {{51, 57}, {66, 45}}, textString = "y")}),
    Documentation(info = "<html><head></head><body><p>This component is a modification of \"fixedTranslation\" of Modelica.Mechanics.Multibody to enable a real time bar changing. 3D visualization was disabled.</p><pre style=\"font-size: 12px;\"><div style=\"font-family: 'MS Shell Dlg 2'; white-space: normal;\"><u>Copyright:</u></div><div style=\"font-family: 'MS Shell Dlg 2'; white-space: normal;\"><br></div><div style=\"font-family: 'MS Shell Dlg 2'; white-space: normal;\"><span style=\"font-family: 'Times New Roman'; orphans: 2; widows: 2;\">Original Work:&nbsp;</span></div><div style=\"font-family: 'MS Shell Dlg 2'; white-space: normal;\"><span style=\"font-family: 'Times New Roman'; orphans: 2; widows: 2;\">Copyright&nbsp;</span><span style=\"font-family: -webkit-standard;\">©</span><span style=\"font-family: 'Times New Roman'; orphans: 2; widows: 2;\">&nbsp;2021-2022 Modelica Association</span></div><div style=\"font-family: 'MS Shell Dlg 2'; white-space: normal;\"><span style=\"font-family: 'Times New Roman'; orphans: 2; widows: 2;\"><br></span></div><div style=\"font-family: 'MS Shell Dlg 2'; white-space: normal;\"><span style=\"font-family: 'Times New Roman'; orphans: 2; widows: 2;\">Derivative Work:&nbsp;</span></div><div style=\"font-family: 'MS Shell Dlg 2'; white-space: normal;\"><p style=\"font-size: 10pt; font-family: Arial, sans-serif;\"><b style=\"font-size: 10pt;\">ShipSIM package</b></p><p style=\"font-size: 10pt; font-family: Arial, sans-serif;\"><em style=\"font-size: 10pt;\">(for license see ShipSIM.UsersGuide.License)</em></p></div></pre>
</body></html>", revisions = "<html><head></head><body><span style=\"font-family: 'MS Shell Dlg 2'; font-size: 12px;\">Rev. 0.0: (01/03/2021): Initial release</span></body></html>"));
end VariableTranslation;

tobolar, 2023-07-17:
Even if helpful in some models, IMO we shall not implement this. The reason especially is that this in principle changes a particular variable form being a parameter to being an input. Doing so, we should also accordingly change 
e.g. `r` and rotation parameters in `Mechanics.MultiBody.Parts.FixedRotation` or `Mechanics.MultiBody.Parts.Fixed`. And so on in further models.

So the rationale behind this change (inc. all known overhead) is not clearly declared and I don't believe we will be able to do so.

HansOlsson, 2023-08-01:
As far as I understand it can also be modeled with 3 translation position components (with exact=true) connected to 3 prismatic joints (I believe I have seen that used somewhere). I don't know whether that is an argument for adding it - or for not adding it.

BTW: The `world` and `import Modelica.SIunits.Conversions.to_unit1;` should just be removed as not needed.

BasilioPV, 2023-08-01:
@HansOlsson You are right, this can be modelled using other MSL components. Mi intention is to crerate a shortcut for use this funcionality. Thanks for point on the not needed lines!