modelica/ModelicaStandardLibrary #4145: Error with frame_resolve math in BasicAbsoluteAngularVelocity sensor

reedsullivan, 2023-06-02:
https://github.com/modelica/ModelicaStandardLibrary/blob/a963d5bac1ac2b6c00ea1ca0567036a17f9f5262/Modelica/Mechanics/MultiBody/Sensors/Internal/BasicAbsoluteAngularVelocity.mo#L23

This function resolves the angular velocity of `frame_a.R` in the world frame, which is then resolved from `frame_a` to `frame_resolve`. That is not correct, as it resolves a world frame angular velocity from `frame_a` to `frame_resolve`.

There are two possible updates, with option 2 being preferred because I believe there are less operations involved.
1. `w = Frames.resolveRelative(Frames.angularVelocity2(frame_a.R), frame_a.R, frame_resolve.R);` where `Frames.angularVelocity1(frame_a.R)` is replaced by `Frames.angularVelocity2(frame_a.R)`.
2. `w = Frames.resolve2(frame_resolve.R, Frames.angularVelocity1(frame_a.R));` where the world frame angular velocity of `frame_a` is resolved in `frame_resolve` directly.

tobolar, 2023-06-12:
It seems you are right, so I clasified this being a bug.

reedsullivan, 2023-06-13:
Looks great! Thanks for the quick fix @tobolar.

PR #4148 (2023-06-15): Fix false calculation of w (omega) when resolveInFrame == frame_resolve
Close #4145 

@beutlich Shall this be backported to v3.2.3 as well?
Oh, I see there is https://github.com/modelica/ModelicaStandardLibrary/wiki/Onboarding-MAP-Lib-Project-Leader-and-Deputy#msl-github-issues-and-pull-requests.
Will try to prepare a proper back-port once this PR is reviewed.

Backport to v3.2.3: #4154
closed 2023-06-15 by MartinOtter