modelica/ModelicaStandardLibrary #4159: M_OLine example PR deleted by mistake!

max-privato, 2023-06-23:
Dear all,
I did create a PR M_OLine example a couple of years ago.
After two years the PR has got life again a few weeks ago.
I was notified of an issue of the PR that I solved just a few hours ago.

So, I was ready to revive that PR successfully.
**Unfortunately I did not realise that deleting a repository from my owh github account I would have deleted everything about that PR.**

So now it seems totally lost.
However, I'm ready to re-create it again, since all the issues of that PR had been solved.
I apologise to anyone that had spent some time on that PR!

I will try to remember as much as I can of the content of the previous discussion on previous PR, and try su create a new PR that is easily checked and, finally and hopefully, merged.

Sorry again for the inconvenience created to anyone!

tobolar, 2023-06-26:
So that PR is no more in the list? https://github.com/modelica/ModelicaStandardLibrary/pulls?q=is%3Apr+max-privato+is%3Aclosed+

ceraolo, 2023-06-28:
I had a strange behaviour on my branch, and I decided... to delete it, under the false assumption that this deletion would not interfere with the already posted PR.
Unfortunately I was mistaken, I found that the old PR #3846 was closed as a consequence of this action.

Therefore I had to create a new PR, starting from my local files.
I got this opportunity to do this under a more respectable name, my surname, "ceraolo", instead of the previously, too naive, "max-privato".

The new PR is #4160.
Sorry for this.