modelica/ModelicaStandardLibrary #4162: IEEE citation style should use American punctuation in UsersGuide.Conventions.UsersGuide.References

gwr69, 2023-06-28:
I just note that the examples given in [UsersGuide.Conventions.UsersGuide.References](https://doc.modelica.org/Modelica%204.0.0/Resources/helpDymola/Modelica_UsersGuide_Conventions_UsersGuide.html#Modelica.UsersGuide.Conventions.UsersGuide.References) use British punctuation conventions—not the appropriate American ones, i.e., the quotation marks should be placed _after_ the comma and not before it (cf. [Wikipedia: Quotation marks in English](https://en.wikipedia.org/wiki/Quotation_marks_in_English)).

Please also see this example for the proper use of punctation in the IEEE Transactions citation style:
https://pitt.libguides.com/citationhelp/ieee

tobolar, 2023-06-28:
New to me, thanks. 'The title of the song was "Gloria," which many already knew.' has really a weird syntax to me. ;-)

gwr69, 2023-06-28:
> New to me, thanks. 'The title of the song was "Gloria," which many already knew.' has really a weird syntax to me. ;-)

Indeed. :)

With an ever increasing number of English language users, but less and less native speakers among them, the idiosyncrasies of British vs. American English may forever be lost—or will happily be mixed:
https://en.wikipedia.org/wiki/Comparison_of_American_and_British_English

HansOlsson, 2023-06-29:
Well, we cannot use that style consistently for all quotes - since parts are documentation where the logical style is preferable.

I tried to look at it and got to the first example Modelica.Blocks.Examples.PID_Controller

> The output of the controller is a torque that drives a motor inertia "inertia1".

Thus for consistency reasons I would propose to not say that we use IEEE style.

henrikt-ma, 2023-06-29:
> > The output of the controller is a torque that drives a motor inertia "inertia1".
> 
> Thus for consistency reasons I would propose to not say that we use IEEE style.

I agree.  To me, the IEEE style, “like this,” is an unreasonable tradeoff between typography hysteria and preservation of grammatical structure.  It was probably more important in the old days of monospaced typewriter text, when it was impossible to snug the comma tightly after a narrow character for the quotation mark, but even back then I can see why the habit wasn't adopted universally.

gwr69, 2023-06-29:
Why should following a citation style necessitate that its use of punctuation needs to transcend to all other uses, e.g., in documentation? Personally, to me as a German, the American use of quotation marks is rather in line with that of my mother tongue, and since I spell like an American most of the time (e.g., oganization, color, humor, center, elevator instead of lift etc.), why not punctuate like most of them do as well?

If you are dropping the explicit mentioning of the IEEE Transactions citation style, I do not find that very consistent. :)

henrikt-ma, 2023-06-29:
> As is so often the case, people frame something as "logical" to make convincing others about a personal choice, which essentially boils down to aesthetics and stylistic _preferences_, an easier task.

In the humanities, I can see this boiling down to aesthetics and stylistic preferences.  However, in our community I believe there are many of us who work closer to parsers, expression trees, and pretty-printers.  In this context, a serialized form where the order doesn't correspond to a traversal of the expression tree is not “logical”.

gwr69, 2023-06-29:
> In the humanities, I can see this boiling down to aesthetics and stylistic preferences. However, in our community I believe there are many of us who work closer to parsers, expression trees, and [...]

You mistake (constructed) programming languages with spoken (natural) languages used by humans. Evolved and spoken language is _different_ from (supposedly strict) mathematical logic.

https://english.stackexchange.com/a/5498/168079 

>Something I find interesting is the marked tendency of programmers to place punctuation outside quotations and parentheses due to a familiarity with the programming convention of "nesting" punctuation.

The different takes on this matter by many style guides (British or American) clearly show, that natural language is everything but a strictly logical programming exercise. :)

gwr69, 2023-06-30:
> Well, we cannot use that style consistently for all quotes - since parts are documentation where the logical style is preferable.
>
> I tried to look at it and got to the first example [...]

@HansOlsson I would agree for your example and in this case even American or typographical punctuation would allow for the above use.

**But let's take a step back:** You mention _consistency_ and with that aim in mind I would suggest that you compare the Modelica Specification (HTML-version) with the documentation for the MSL (which is in itself not consistent in the use of typography as far as I can see). Also have a look at the current web documentations for MATLAB, Maple, Wolfram Mathematica, GitHub, Microsoft .NET and other technical systems. Please tell me, how many quotation marks you will find in these documents, which will have comparable tasks and challenges as the MSL documentation?

There is **almost no use of quotation marks in modern technical documentations** on the web! The reason for this is that when text is separated from surrounding text by its styling (e.g., color, italics, non proportional fonts), quotation marks are not needed (neither in the UK nor the US)! It seems that the MSL documentation stems from a time, where the "common denominator" for computer text output in a browser was indeed Henrik's "monospaced typewriter text."

Why should the documentation for the MSL look different from the Modelica Specification or modern technical documentation?