modelica/ModelicaStandardLibrary #4166: Modelica Standard Library should encourage following either British or American English consistently

gwr69, 2023-07-03:
Extending #4162 I would suggest that the Modelica Standard Library should—if for historical reasons it cannot do so itself—encourage **consistent use of either British (BE) or American English (AE)**.

Within the MSL, it seems that the use of AE (mostly) prevails:

- Dates are written as "July 31, 2023" (AE) not "31 July 2023" (BE)
- There are 10 matches for "organization" (AE) versus 1 match for "organisation" (BE)
- There are 19890 matcher for "color" (AE) versus 1 match for "colour" (BE)
- There are 286 matches for "center" (AE) versus 0 matches for "centre" (BE)
- There are 148 matches for "modeling" (AE) versus 72 matches for "modelling" (BE)
- There are 4 matches for "reorganisation" (AE) versus 1 match for "reorganisation" (BE)
- There are 108 matches for "behavior" (AE) versus 83 matches for "behaviour" (BE)
- There are 13 matches for "meter" (AE) versus 20 matches for "metre" (BE)
- There are 0 matches for "kilometre" (AE) versus 9 matches for "kilometre" (BE)
- There are 12 matches for "program" (AE) versus 1 match for "programme" (BE)
- There are 463 matches for "i.e.," (AE) versus 57 matches for "i.e. " (BE)
- There are 433 matches for "e.g.," (AE) versus 66 matches for "e.g. " (BE)
- ...

Without (once more) getting into petty arguments regarding "parsing of syntax" (which appears to be rather [nontrivial](https://en.wikipedia.org/wiki/Sentence_processing) for humans), I believe one should encourage consistent use of either AE or BE, where usage is clearly different, e.g., "organization" is correct—albeit infrequent—use in BE.

beutlich, 2023-07-03:
See also #2750 which was closed years ago as wontfix.

gwr69, 2023-07-04:
> See also #2750 which was closed years ago as wontfix.

Thanks for pointing me to that issue. The argumentation there does not quite convince me and it also does not completely apply to what I am calling for here.

With regard to the use of language, I would argue against incorporating an attitude in line with Paul Feyerabend's famous verdict, "Anything goes!" There is a reason for most academic journals or journalistic publications adhering to some kind of [**style guide**](https://en.wikipedia.org/wiki/Style_guide):

>"A style guide establishes standard [style](https://en.wiktionary.org/wiki/style#Noun) requirements to improve [communication](https://en.wikipedia.org/wiki/Communication) by ensuring [consistency](https://en.wiktionary.org/wiki/consistency#Noun) both within a document, and across multiple documents. Because practices vary, a style guide may set out standards to be used in areas such as [punctuation](https://en.wikipedia.org/wiki/Punctuation), [capitalization](https://en.wikipedia.org/wiki/Capitalization), [citing sources](https://en.wikipedia.org/wiki/Citing_sources), formatting of numbers and dates, [table](https://en.wikipedia.org/wiki/Table_(information)) appearance and other areas. The style guide may require certain [best practices](https://en.wikipedia.org/wiki/Best_practice) in [writing style](https://en.wikipedia.org/wiki/Writing_style), [usage](https://en.wikipedia.org/wiki/Usage), [language composition](https://en.wikipedia.org/wiki/Composition_(language)), [visual composition](https://en.wikipedia.org/wiki/Composition_(visual_arts)), [orthography](https://en.wikipedia.org/wiki/Orthography), and [typography](https://en.wikipedia.org/wiki/Typography)." (Source: Wikipedia)

Likely, the rather _inconsistent_ use within the MSL documentation should have been avoided or at least ameliorated, if clearer guidance had been available—given the multiplicity of authors from different regions of the world?

While recourse to Google Ngrams is interesting, it falls short with regard to authoritative guidance on _consistent_ use of language, i.e., you will look up individual words _independently_, when you should have looked up a _joint probability distribution_. With regard to _spelling_, lectors around the world would probably not turn to Google, but use a current edition of a well known dictionary.

Looking up "behavior" vs. "behaviour" in [_Merriam-Webster_](https://www.merriam-webster.com/dictionary/behaviour), the [_American Heritage Dictionary_](https://ahdictionary.com/word/search.html?q=Behaviour&submit.x=0&submit.y=0), [_Chambers Dictionary_](https://chambers.co.uk/search/?query=behavior&title=21st), or [_Collins Dictionary_](https://www.collinsdictionary.com/dictionary/english/behavior) will unsurprisingly and _unambiguously_ tell you that "behaviour" is chiefly British, while "behavior" clearly is the North American spelling.

Consulting these dictionaries will also leave not doubt about "_color/clour_", "_meter/metre_", "_gray/grey_", "_organization/organisation_" etc. The latter is a more flexible example, but often flexibility is _not symmetrical_, e.g., you will only find "_homoeostasis_" in  British English—in AE it is "_homeostasis_", only.

There are certainly many variants of the English language, as the following table shows ([Source](https://en.wikipedia.org/wiki/Comparison_of_American_and_British_English#/media/File:International_English_Spelling_updated.svg)):

[image]

But there should be no doubt about British and American English being the two most important variants—unless you are decidedly publishing in Canada or Australia. Nevertheless, texts edited in Vancouver, Chicago, London, or Adelaide will have one thing in common: Once they have decided for a certain language variant or style (e.g., a column in the table), they will stick to their choice and not have "metre" and "meter" appear in the same text.

While the MSL documentation may remain an unfortunate case of `wontfix`, I am arguing for learning from the experience and for **providing a recommendation for Modelica authors to stick to one style and to be consistent**.

For all those, who argue against the predominant use of American English in the MSL, consulting the table above will maybe call for renaming _Modelica_ to **_Modellica_**? ;-)