modelica/ModelicaStandardLibrary #4168: Rename unit with historical meaning of Debye

henrikt-ma, 2023-07-07:
When adding support in System Modeler for all units seen in the MSL, we encountered a definition of `"debye"` that [Wikipedia](https://en.wikipedia.org/wiki/Debye) refers to as "historical":
https://github.com/modelica/ModelicaStandardLibrary/blob/e55884bab622a7393af9234958c218ef5984fc70/Modelica/Media/package.mo#L758

Perhaps the unit should be renamed somehow to not cause confusion in relation to the (assumedly) more modern CGS definition of the Debye?  To make a concrete proposal, how about using the unit symbol `"debyeCharge"` instead of `"debye"` in `DipoleMoment`:
https://github.com/modelica/ModelicaStandardLibrary/blob/e55884bab622a7393af9234958c218ef5984fc70/Modelica/Media/package.mo#L6631-L6635

The reason behind suggesting the name `"debyeCharge"` is that the definition corresponds to the charge interpretation of the statcoloumb:
https://en.wikipedia.org/wiki/Statcoulomb

henrikt-ma, 2023-07-07:
Related to the "debye of charge", what _exactly_ is the intended relation between this unit and `"C.m"`, considering the updated status of `mu_0` and `epsilon_0` (compare https://github.com/modelica/ModelicaStandardLibrary/pull/4161 and https://github.com/modelica/ModelicaStandardLibrary/pull/4163)?

With 10 c = 2997924580 m/s, should the value of 1 debye of charge be (1 C m)/(10 c) or the solution `q` to the equation
```
1 dyne = q^2 / (4π epsilon_0 (1cm)^2)
```
where `epsilon_0` = `1 / (mu_0 c^2)`? The former alternative corresponds to setting `mu_0` = 4π 10^(-7) N/A^2, but this is not the value we use.

HansOlsson, 2024-05-15:
I don't think we should change the name debye as it is well-known, and calling it a "charge" seems confusing as it is a dipole moment - i.e., charge times distance (summed).
Statcoloumb is, of course, a charge, but so is the normal coloumb; for SI-definition of dipole unit.
If we were to change it the abbreviation "D" seems simplest - but you never know if SI will define that in the future.

I don't think we need to be too careful in the definition of dipole; so the simple idea is to use
1 D = 1e-21 Cm2/s / (299792558 m/s).

We also need to clean up:
-  `3.33564e10-30 C.m` it should be `3.33564e-30 C.m`
- Possibly the values:
  - CO has 0.1 D but links give it as 0.112 D or 0.122 D
  - NH3 has 1.5 but links give it as 1.42 D
  - H20 has 1.8 D but links give it as 1.85 D
- And clearly the max-value should be removed, it is given as 2, but KBr has a dipole moment of 10.41 D.

https://en.wikipedia.org/wiki/Dipole#Molecular_dipoles
https://en.wikipedia.org/wiki/Carbon_monoxide
https://en.wikipedia.org/wiki/Debye

HansOlsson, 2024-05-15:
And "historical" on the wikipedia page is a bit misleading, as there wasn't a prefixed SI-unit in a reasonable range until 2022; https://github.com/modelica/ModelicaSpecification/pull/3507

I'm convinced that debye will remain in use for many years; until 1 qC.m=0.3 D might take over.