modelica/ModelicaStandardLibrary #4178: Recompilation of binaries necessary

MatthiasBSchaefer, 2023-08-10:
In the master branch e.g. commit ce36c07 examples which include a CombiTimeTable are not compiling, because the function
ModelicaStandardTables_CombiTimeTable_init3
can not be found in the binaries. The function was added to ModelicaStandardTables.c after the last compilation of the binaries. 
So a recompilation of the binaries for all platforms is necessary.

beutlich, 2023-08-10:
I know, this is done (usually by me) just before a new release of the MSL is tagged.

beutlich, 2023-08-10:
Duplicate of #3911.

beutlich, 2023-08-10:
I am actually waiting for #4047 to get merged first. ⏳

beutlich, 2023-08-10:
Meanwhile you can find the updated Windows binaries in https://github.com/beutlich/ModelicaStandardLibrary/tree/update-binaries.

beutlich, 2024-03-13:
Resolved by #4250.

PR #4250 (2024-03-13): Rebuild binaries for Windows (MSVC, TDM-GCC) and Linux (GCC)
I rebuilt the binaries using

* MSVC VS 2005
* GCC (TDM64-1) 5.1.0
* Linux GCC 4.8.5

to fix #4178.
closed 2024-03-13 by beutlich