modelica/ModelicaStandardLibrary #4185: Over-zealous null-pointer guards in ModelicaStandardTables_CombiTimeTable_init (etc)

HansOlsson, 2023-08-31:
If you look at the declaration of ModelicaStandardTables_CombiTimeTable_init (and similar ones) you notice that all pointer arguments are declared to be non-NULL (using nonnull-attribute for Gcc and SAL for Visual Studio).

However, if the table is read from a file then table isn't used (or even checked) and can safely be NULL.

The correction for Gcc will be a bit messy (specify all non-NULL pointers) so I didn't want to make the PR unless it is likely to be accepted.

beutlich, 2023-09-04:
Even if the table is read from file, it still gets passed as `double*` pointer and cannot be null if called from a Modelica environment. If called from a non-Modelica environment, well, it can be different.

HansOlsson, 2023-09-05:
> Even if the table is read from file, it still gets passed as non-null `double*` pointer if called from a Modelica environment. Having a Modelica environment was the initial assumption though. If called from a non-Modelica environment, well, it can be different and table can be null if it should be read from file.

I assume you mean "Modelica environment" as short for "Modelica Standard Library environment", since we have some that use the C-sources directly in their Modelica code - bypassing/copying the MSL models. (In particular that's how it was found.)

beutlich, 2023-09-05:
By "Modelica env" I meant a Modelica simulation environment as Dymola or SimulationX. In Modelica Language it is not possible to pass a null pointer to an external function or external object ctor.

beutlich, 2024-01-13:
An idea would be to have to different ctors of the external table objects - one for construction of table from file and other for construction from passed array.