modelica/ModelicaStandardLibrary #4186: Class name as text in Diagram in Electrical.Analog

maltelenz, 2023-08-31:
A lot of the examples in `Modelica.Electrical.Analog.Examples` have a large `Text` item on the `Diagram` basically replicating the information of the class name.

Other sub-libraries in MSL don't have this, and any tool would presumably show the class name somewhere whenever showing the diagram, so replicating this information on the diagram only adds clutter.

Would a pull request removing these `Text` items be accepted?

HansOlsson, 2023-09-06:
It is also inconsistently done in that package; which makes it less useful.

However, note that even if Modelica.Electrical.Analog.Examples.ResonanceCircuits has texts in a similar style I think they should be kept (and such texts should be encouraged), whereas Modelica.Electrical.Analog.Examples.ShowVariableResistor has a text that is roughly the class name.

maltelenz, 2023-09-06:
> It is also inconsistently done in that package; which makes it less useful.
>
> However, note that even if Modelica.Electrical.Analog.Examples.ResonanceCircuits has texts in a similar style I think [...]

I agree with everything you say here. There are definitely cases where text on diagrams can be used for good things, like in `ResonanceCircuits`. I would not want to remove those cases.