modelica/ModelicaStandardLibrary #4192: Unit errors when calling Modelica.Mechanics.MultiBody.Frames.from_nxy

qlambert-pro, 2023-09-18:
`Modelica.Mechanics.MultiBody.Frames.from_nxy` is expecting unit `"1"` inputs but in at least:
`Modelica.Mechanics.MultiBody.Parts.BodyCylinder` and `Modelica.Mechanics.MultiBody.Parts.BodyBox` it is being called with variables expressed in `"m"`.

I see two solutions to address this issue, one would be to divide the corresponding variable by its `length`, and the other would be to call `Modelica.Units.Conversions.to_unit1`.

Which one is preferred?

HansOlsson, 2023-09-18:
I see a third option: remove unit="1" as it doesn't seem to make sense.
Will investigate.

henrikt-ma, 2023-09-18:
> I see a third option: remove unit="1" as it doesn't seem to make sense.
> Will investigate.

What is nice about having `unit = "1"` on both inputs is that it ensures that both inputs have the same unit.  It's a bit weird in my opinion to call this function with incompatible units on the two inputs.

HansOlsson, 2023-09-18:
It was introduced as part of https://github.com/modelica/ModelicaStandardLibrary/commit/01a89e871453b10be85f78b7c39d3c0cc4e3872f with the intent of having unit="1" on unit-vectors (which is good in general), and many functions were improved by that - but some were messed up like this one.

However, `from_nxy` isn't restricted to unit-vectors - as it takes any vectors and normalizes them before using them, and thus shouldn't have unit on the inputs.

So, remove unit from `from_nxy` and instead have it on the result of `Modelica.Math.Vectors.normalize`.
(And try to figure out why normalize uses Modelica.Constants.eps ...)
I believe Modelica.Mechanics.MultiBody.Visualizers.Internal.* has the same issue for n_x and n_y as well as Modelica.Utilities.Internal.PartialModelicaServices.Animation.PartialShape and ModelicaServices.Animation.Shape.

PR #4194 (2023-11-14): Correct units for normalizing vectors.
Closes #4192
The basic idea is that non-normalized vectors don't need unit.
closed 2023-11-14 by MartinOtter