modelica/ModelicaStandardLibrary #4197: Unit issue in Modelica.Mechanics.Translational.Components.ElastoGap

HansOlsson, 2023-09-19:
The commit https://github.com/modelica/ModelicaStandardLibrary/commit/0315615184538fe055cc1c941219cb524c3db9ca changed ElastoGap to be unit-unsafe. In particular the previous `s_ref` parameter was removed, and the non-linear exponent `n=0` was replaced with the general exponent `n=1` (roughly it replaced `(s_rel-s_rel0)*abs((s_rel-s_rel0)/s_ref)^n` by `(s_rel-s_rel0)^n`).

The solution seems to be to restore `s_ref` (e.g., `parameter SI.Position s_ref=1 annotation(enable=n>1 or n<1);`) and replace` c*abs(s_rel - s_rel0)^n` by `c*abs((s_rel - s_rel0)/s_ref)^n*s_ref`. The extra factor `s_ref` ensures that `c` keeps its previous unit (and especially doesn't have a unit depending on the real parameter `n`).

My assumption is that if someone is investigating non-linear elasto gaps the additional freedom of setting `s_ref` is actually helpful.

henrikt-ma, 2023-09-20:
```
parameter SI.Position s_ref=1 annotation(enable=n>1 or n<1);) 
```
Is conditioning on the typically non-evaluated, `Real`-valued, `n` such a good idea?  Isn't it better to use the description of `s_ref` to say that it only matters for n != 1?

> The solution seems to be to restore `s_ref` (e.g., `parameter SI.Position s_ref=1 annotation(enable=n>1 or n<1);`) and replace` c*abs(s_rel - s_rel0)^n` by `c*abs((s_rel - s_rel0)/s_ref)^n*s_ref`. [...]

While multiplying with `s_ref` would "restore" the unit, wouldn't it result in an unexpected scaling effect for other values than 1?

tobolar, 2023-09-21:
What about to introduce a unit conversion like:
```
constant SI.Length convUnitLenght = 1;
Real e_del;

e_del = (s_rel - s_rel0)/convUnitLenght;
f_c = smooth(1, noEvent(if contact then -c*convUnitLenght*abs(e_del)^n else 0));
```

HansOlsson, 2023-09-21:
> What about to introduce a unit conversion like:
> 
> ```
> constant SI.Length convUnitLenght = 1;
> Real e_delta;
> 
> e_delta = (s_rel - s_rel0)/convUnitLenght;
> f_c = smooth(1, noEvent(if contact then -c*convUnitLenght*abs(e_delta)^n else 0));
> ```

That would also work, but I thought exposing it as a parameter (as it was a long time ago) would be more useful; for the non-linear case.

tobolar, 2023-09-21:
@HansOlsson  You mean `s_ref`? 

I agree with @henrikt-ma 
> While multiplying with s_ref would "restore" the unit, wouldn't it result in an unexpected scaling effect for other values than 1?

and referring the model documentation, this wouldn't not the intended behaviour, right?

tobolar, 2023-09-25:
@HansOlsson I like your approach. It simply changes `f = c * delta_x^n` to `f = f_ref * k^n`.  Consequently,
- `cNew` shall better be called like `f_ref` since it is indeed a force. Adding some examples in the documentation would be helpful - I can care about.

Regarding current temporary clean solution:
- shouldn't be `cNew = c*s_ref^n` better?
- why not `SI.TranslationalSpringConstant c`?

We can optionally put `cNew` and `s_ref` into "Advanced" tab for now.

HansOlsson, 2023-09-25:
> @HansOlsson I like your approach. It simply changes `f = c * delta_x^n` to `f = f_ref * k^n`. Consequently,
> 
> * `cNew` shall better be called like `f_ref` since it is indeed a force. Adding some examples in the documentation would be helpful - I can care about.

Good.

> Regarding current temporary clean solution:
> 
> * shouldn't be `cNew = c*s_ref^n` better?

Only `f_ref = c*s_ref` is unit-correct with current c. I agree that the other formula would be better, except for the compatibility.

> * why not `SI.TranslationalSpringConstant c`?

Yes, I just didn't remember that name.

> We can optionally put `cNew` and `s_ref` into "Advanced" tab for now.

Good enough.

tobolar, 2023-09-29:
Regarding PR #4200, we should clarify the signs of `f_ref` and `s_ref`.
The point is that `f_c` is negative for `contact = true`, i.e. when `s_rel < s_rel0`. This means, the contact occures when the difference `s_rel-s_rel0` is negative. 
IMO this implies that both `f_ref` and `s_ref` shall be limited by `max = 0`. But may be this is too confusing for users.

PR #4200 (2023-12-21): Improve ElastoGap
Closes #4197 
(See that ticket for the discussion.)
commit 043aaab9 (2023-12-21): Improve ElastoGap
Closes #4197

Prefer code to pre, and move things.

Add scaling ratio variable and add parameter description

Simplify computation of f_d

Documentation: add non-braking spaces where needed

Add scaling ratio variable and add parameter description

Simplify computation of f_d

Documentation: add non-braking spaces where needed

Improve documentation concerning s_rel and f_rel

Enable only positive values of s_rel and f_rel
closed 2023-12-21 by HansOlsson