modelica/ModelicaStandardLibrary #4201: Utilities.Streams.writeRealMatrix seems to not work properly

mpradovelasco, 2023-09-29:
Hi, 

This issue derives from a previous comment in discussion forum (#4198). I have a block component that writes in a mat file using the following code:

``` Modelica
String dataName(start = getInstanceName() + "_dsignal", fixed = true);
algorithm  
  /* Write */
  when terminal() and saveToFile then
    dataName := Utilities.Strings.replace(dataName, ".", "_");
    success := Utilities.Streams.writeRealMatrix(fileName, dataName, 
      dataSignal, format=format, append=true);
  end when;
```
The first line simply substitutes "." by "_" for matlab compatibility. 

Despite using append = true when calling, the data are written only if the file it does not exist. This manner, if I have two components that append data and the file does not exist, only the first call is attended (the names of arrays to be saved are different). The following image shows the file with the data from the first call. 

[image]

If the file exist, no one writes in the file (in that case Matlab indicates that mat file is empty). I have tried with different Mat formats and append values without success.

This issue occurs with System: MSL 4.0.0 under OMEdit 1.22 and Windows 11.

Thanks for your help.

beutlich, 2023-10-02:
Since the code snippet misses the declarations and assignments/equations for saveToFile, fileName, dataSignal and format, can you please provide a [SSCCE](http://sscce.org/) for reproduction. Thanks.

mpradovelasco, 2023-10-03:
Yes, the following simplified code is able to reproduce the issue.

``` Modelica
package StreamsWritting
  import Modelica.Utilities;
  block pSaveSimp
    parameter String format = "7" "MATLAB MAT file version: \"4\" -> v4, \"6\" -> v6, \"7\" -> v7" 
      annotation(
      Dialog(enable = true),
      choices(choice = "4" "MATLAB v4 MAT file", 
        choice = "6" "MATLAB v6 MAT file", 
        choice = "7" "MATLAB v7 MAT file"));
    parameter String fileName = "default.mat" "Name of the file" 
      annotation(
      Dialog(enable = true, saveSelector(filter = "Mat files (*.mat)", 
        caption = "Define mat file for writting")));
    parameter Integer nSamples(min = 1) = 3 "Number of samples to store";
    parameter Integer nin = 1;
    Real dataSignal1[nin, nSamples] "Read data";
    Real dataSignal2[nin, nSamples] "Read data";
    String dataName1(start = getInstanceName() + "_dsignal1", fixed = true) "Name of data variable to save in file";
    String dataName2(start = getInstanceName() + "_dsignal2", fixed = true) "Name of data variable to save in file";
  protected
    output Boolean success(start = true, fixed = true) "= true if writing to .mat is successful";
  equation 
    for i,j loop
      dataSignal1[i,j] = i+j;
      dataSignal2[i,j] = i+exp(j);
    end for;
  algorithm
    /* Write */
    when terminal() then
      dataName1 := Utilities.Strings.replace(dataName1, ".", "_");
      dataName2 := Utilities.Strings.replace(dataName2, ".", "_");
      success := Utilities.Streams.writeRealMatrix(fileName, dataName1, 
        dataSignal1, format = format, append = true);
      success := Utilities.Streams.writeRealMatrix(fileName, dataName2, 
        dataSignal2, format = format, append = true);
    end when;
    annotation(
      Icon(graphics = {Rectangle(origin = {1, 3}, extent = {{-79, 57}, {79, -57}})}));
  end pSaveSimp;
  block UsopSave
    pSaveSimp pSaveSimp1(fileName = "C:/DESARROLLOYDATOS/Areas/ProyectosNoFinanciados/CybSim/simExp/CybSim.test.StreamsWritting.UsopSave/SaveData.mat")
      annotation(
      Placement(transformation(origin = {-6, 20}, extent = {{-10, -10}, {10, 10}})));
  end UsopSave;
end StreamsWritting;
```
I define the fileName to the desired location, as shown in the dialog of the figure:

[image]

If I execute the mode when the file does not exist, then the file is created with the first array that is written, but the second array is not written despite append = true in the writeRealMatrix argument. The following image shows the datafile on Matlab:

[image]

After a second execution of the model with the existing file, nothing is written and previous data dissapears.

[image]

I attach the datafile: 
[SaveData.mat.txt](https://github.com/modelica/ModelicaStandardLibrary/files/12789215/SaveData.mat.txt)

Tests performed in OpenModelica 1.23-dev and Windows 11.

Thank you for your help!

beutlich, 2023-10-03:
Seems like a tool issue to me since I cannot reproduce with Dymola 2023. The SaveData.mat file is written as expected.

@HansOlsson I noticed a different issue with Dymola (but not with OMC) though: The two commands of `Utilities.Strings.replace` are (basically) not executed and hence, generate invalid MATLAB variable names.

```
$ ./matdump.exe -f whos SaveData.mat
Name                       Size           Bytes          Class

UsopSave.pSaveSimp1_dsignal1       1x3                  24  mxDOUBLE_CLASS
UsopSave.pSaveSimp1_dsignal2       1x3                  24  mxDOUBLE_CLASS
```

HansOlsson, 2023-10-03:
> @HansOlsson I noticed a different issue with Dymola (but not with OMC) though: The two commands of `Utilities.Strings.replace` are (basically) not executed and hence, generate invalid MATLAB variable names (and MAT-files).

As far I can see a work-around for that is to just remove the start-values for the strings and use:

 ```
when terminal() then
    dataName1 := getInstanceName() + "_dsignal1";
    dataName2 := getInstanceName() + "_dsignal2";
    dataName1 := Utilities.Strings.replace(dataName1, ".", "_");
    dataName2 := Utilities.Strings.replace(dataName2, ".", "_");
```
(I actually find that clearer than having a separate start-value. Will see which version it will be corrected for.)

beutlich, 2024-01-13:
Clsoing as too-issue(s).

closed 2024-01-13 by beutlich