modelica/ModelicaStandardLibrary #4209: Problem with the function setState_phX

congtoantran, 2023-10-20:
I have a library which works well on Modelica 3.3.2, but when converted to Modelica 4.0.0, it doesn't work anymore, and the message error is "Non-real equation in continuous time are not legal:". 

I'm trying to understand the problem by using a very simple example (see below): 

model model1a
  replaceable package Medium = Modelica.Media.R134a.R134a_ph "Medium model" 
    annotation (choicesAllMatching=true);
  Medium.ThermodynamicState state_a ;
  Modelica.Units.SI.AbsolutePressure p=Medium.p_default;
  Modelica.Units.SI.SpecificEnthalpy h=Medium.h_default;

equation 
  state_a = Medium.setState_phX(Medium.p_default, Medium.h_default);
  //     state_a = Medium.setState_phX(p, h);
end model1a;

The above model runs sucessfully. But when I redefine the variable state_a by using the second line in the equation part, the model does not work and the error is : 
  Non-real equation in continuous time are not legal:
  state_a.phase = Modelica.Media.R134a.R134a_ph.getPhase_ph(p, h);

Any help to understand the error? Many thanks. I use Dymola 2021, Modelica 4.0.0

beutlich, 2024-01-13:
@HansOlsson I can reproduce in Dymola 2024x, but not in SimulationX. Is it a tool issue?

It works though (in both tools) if the variability is changed to `parameter`.

```mo
model Model1b
  replaceable package Medium = Modelica.Media.R134a.R134a_ph "Medium model"
    annotation (choicesAllMatching=true);
  Medium.ThermodynamicState state_a;
  parameter Modelica.Units.SI.AbsolutePressure p=Medium.p_default;
  parameter Modelica.Units.SI.SpecificEnthalpy h=Medium.h_default;

equation 
  //state_a = Medium.setState_phX(Medium.p_default, Medium.h_default);
  state_a = Medium.setState_phX(p, h);
end Model1b;
```

HansOlsson, 2024-01-15:
I would say it is a library issue that can be corrected.

The issue is that this record has a Boolean member (phase), and a Boolean equation requires that both sides are discrete-time valued, section 3.8.4.

The correction would be that the phase-functions get inlined with events.

casella, 2024-07-02:
I checked @HansOlsson's pull request and it looks good to me.

@arunkumar-narasimhan please back-port this to maint/4.1.0

casella, 2024-07-02:
This issue is actually a regression from v.4.0.0 but I also marked it with the 4.1.0-dev label so we don't forget about it

casella, 2025-02-11:
@Esther-Devakirubai ping, please back-port this to maint/v4.1.x

casella, 2025-02-24:
Once we get a second review on #4541 this will be closed.

maltelenz, 2025-02-25:
This introduces a new model in ModelicaTest, so this issue should only be closed once we have new reference results for that model.

casella, 2025-02-25:
LTX needs to add reference results for this test. @MatthiasBSchaefer will take care of that. Then, we'll need to back-port it.

casella, 2025-03-08:
> This introduces a new model in ModelicaTest, so this issue should only be closed once we have new reference results for that model.

Here they are: https://github.com/modelica/MAP-LIB_ReferenceResults/tree/v4.1.0/ModelicaTest/Media/TestOnly/R134a_setState_phX

PR #4263 (2024-07-02): Generate events in functions; to avoid non-discrete booleans.
Closes #4209

I don't know if significant enough to push for release.
commit 54f4a7a0 (2024-07-02): Generate events in functions; to avoid non-discrete booleans. (#4263)

* Generate events in functions; to avoid non-discrete booleans.
Closes #4209

* Apply suggestions from code review

Fix annotations.

* Need Inline as well (at least for Dymola - and shouldn't be problem).

* Add test-case

* Update ModelicaTest/Media.mo

* Inlining the function as suggested.

---------
commit 0f769acf (2025-02-25): Generate events in functions; to avoid non-discrete booleans. (#4263) (#4541)

* Generate events in functions; to avoid non-discrete booleans.
Closes #4209

* Apply suggestions from code review

Fix annotations.

* Need Inline as well (at least for Dymola - and shouldn't be problem).

* Add test-case

* Update ModelicaTest/Media.mo

* Inlining the function as suggested.

---------
closed 2025-03-08 by casella