modelica/ModelicaStandardLibrary #4214: Broken link to 3-Clause BSD license on modelica.org

tobolar, 2023-10-24:
The link to the 3-Clause BSD license on https://modelica.org/licenses/modelica-3-clause-bsd seems to be broken (after the webpage relaunch):
https://github.com/modelica/ModelicaStandardLibrary/blob/d3ad60c312774a1bf84a74e830ccd55abcfb09d2/Modelica/package.mo#L2829

HansOlsson, 2023-10-24:
I think we need to decide on https://github.com/modelica/www.modelica.org/issues/45 first
A benefit of keeping the existing license-part of the web-site is that existing links for generated documentation will remain useful.

tobolar, 2023-10-24:
@HansOlsson Sorry, I was not aware of it.

beutlich, 2024-01-14:
As an idea: Should we refer to https://web.archive.org/web/20220930065028/https://modelica.org/licenses/modelica-3-clause-bsd ?

beutlich, 2024-05-27:
Issue is resolved. Thanks @t-sommer !

closed 2024-05-27 by beutlich