modelica/ModelicaStandardLibrary #4232: Mismatching order in pacakge.order

eshmoylova, 2023-11-15:
In [Directory Hierarchy Mapping section of the Specification](https://specification.modelica.org/maint/3.6/packages.html#directory-hierarchy-mapping), it says:

> Classes and constants that are stored in package.mo are also present in package.order but their relative order should be identical to the one in package.mo (this ensures that the relative order between classes and constants stored in different ways is preserved).

There are two cases where the order of classes in package.mo and package.order do not match:
[Media/IdealGases/Common/package.mo](https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Media/IdealGases/Common/package.mo) does not match [Media/IdealGases/Common/package.mo](https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Media/IdealGases/Common/package.order), in particular, DataRecord is not followed by SingleGasNasa in package.order

[Math/package.mo](https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Math/package.mo) does not match [Math/package.order](https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Math/package.order): classes defined outside of package.mo are interspersed with classes defined in package.mo.

If the the order given in package.order is preferred, the classes in package.mo need to be taken out into stand-alone files.

HansOlsson, 2023-11-15:
> In [Directory Hierarchy Mapping section of the Specification](https://specification.modelica.org/maint/3.6/packages.html#directory-hierarchy-mapping), it says:
>
>
> There are two cases where the order [...]

I don't know if the text is unclear, some misunderstanding, or the variants have diverged but what I see is that
package.order contains

- **DataRecord**
- Functions
- **SingleGasNasa**
- **MixtureGasNasa**
- FluidData
- SingleGasData

package.mo contains the ones in bold in the following order:
- **DataRecord**
- **SingleGasNasa**
- **MixtureGasNasa**

So, the sub-set of classes in package.mo keep their relative order from package.order; the other classes Functions, FluidData, SingleGasesData are defined in other files. This allows you to freely order all of the classes (and constants) - regardless of whether they are stored in package.mo or in a separate file (or directory); and without having the weird situation that you have one order in package.mo and then shuffle them in package.order.

> [Math/package.mo](https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Math/package.mo) does not match [...]

I believe it is similar here, but didn't go through all of them.

eshmoylova, 2024-04-11:
Closing. The Language group agreement was that this should be allowed.

closed 2024-04-11 by eshmoylova