modelica/ModelicaStandardLibrary #4233: Old value for universal gas constant used for calculation of specific gas constant

Parti-Gyle, 2023-11-19:
I noticed that the specific gas constant stated in the data records for the ideal gases (`SingleGasesData.mo`), has been calculated with a universal gas constant value of $`R=8.3144720`$; which I believe would have been the value for $`R`$ when the Media library was written. 

However, in the McBride B.J., Zehe M.J., and Gordon S. (2002) (*NASA Glenn coefficients*) paper they use a value $`R=8.314510`$ (Appendix A). Yet since then, the universal gas constant has been redefined $`R=N_A.k=8.31446261815324\; \: J\cdot K^{-1}\cdot mol^{-1}`$.

I know we're talking about a difference in the 5th/6th significant figure, but should all the specific gas constants be recalculated `$R_S`$? And if so, what value of the universal gas constant $`R`$ should be used?

I'm happy to submit a pull request the the relevant changes if it's deemed necessary.

beutlich, 2023-11-20:
Thanks for raising this issue.

I am not sure if we really need the calculated value for R_s, or rather can live with an expression `final R_s=Modelica.Constants.R/MM`.

Parti-Gyle, 2023-11-20:
I agree that would remove the need to update all the specific gas constants whenever the universal gas constant in `Modelica.Constants.mo` gets updated.  However, $R$ in MSL is a different value to that which was used in the NASA paper and this difference will propagate to the thermodynamic properties calculated with the NASA Glenn coefficients.  

I know the error will be small, using a newer value of $R$, but the thermodynamic properties calculated will differ from what the original NASA model produces.

Should the specific gas constants not be recalculated with `$R=8.314510`$, to remain true to the NASA source?

mestinso, 2023-11-20:
I did a cursory glance here and based on my experience with fluid property routines such as this one, yes normally one should always use the universal gas constant that was used by the authors of the routine. In this case, that appears to be the 8.314510 value.

So, the R_S value in each data record should probably be replaced by MM/R_NASA2002, where R_NASA2002=8.314510.

...moving forward, it will be nice when all routines use the now truly constant new universal gas constant value. But for all routine created before ~2020, yes, I think we will have these other/older gas constant values in the mix. (Note that the alternative is much messier).

With that said, probably best to get agreement/confirmation from the author/officer for this portion of the library.

hubertus65, 2024-01-08:
Thanks Matthis, for the link, quite insightful. I don't find this easy, trivial to decide.

- Exact reproduction of the NASA-Values would require to keep the old value, as in the pull request. 
- Different values for different gases give more trouble than minor deviations, as @thorade showed via the link
- The latest value of R is now final and a constant that won't change independent of measurements. 
- Within the current MSL we only have a single R, but Modelica is open enough that it could happen. 
- Engineering decisions rarely hinge on the value of the 5th significant digit. Definitely not in Thermodynamic questions
- Given that NIST is one of the authorities in fluid property calculation, I would tend to follow theri lead, even though we currently don't have the same issue. 

My feeling is that it would be best to: update the R-value to the current final value and live with the (very minor) differences that gives for the original NASA data. I'm happy to hear further opinions.

casella, 2024-01-17:
See discussion in #4238

closed 2024-01-18 by Harisankar-Allimangalath