modelica/ModelicaStandardLibrary #4234: Extracting variable from conditional models

GarronFish, 2023-11-21:
In the example below there are two similar conditionally declared models (i.e. contitionalModel1 and conditionalModel2 that contain similar internal variables, the user would like to get infomation of an arbitrary variable from the model that is conditionally declared.

`model ExtractVariableFromConditionalModelEx
  parameter Boolean enableModel1Not2 = true;

  Modelica.Blocks.Sources.Constant conditionalModel1(k=1) if enableModel1Not2;
  Modelica.Blocks.Sources.Constant conditionalModel2(k=2)
    if not enableModel1Not2;
  output Real y1 "k from the conditionally declared model (i.e. either conditionalModel1 or conditionalModel2";
// What is the correct way to connect y1 to k in the conditionally declared models?  
    
    
end ExtractVariableFromConditionalModelEx;
`

setting y1 = if enableModel1Not2 then conditionalModel1.k else conditionalModel2.k, is not supported and the following error is generated (using Dymola 2024x Beta3 with Advanced.Modelica.CheckConditionallyEnabled = true):

_Undeclared variable: conditionalModel2.k since the declaration of conditionalModel2 was conditionally removed
in the definition equation
y1 = (if enableModel1Not2 then conditionalModel1.k else conditionalModel2.k)_

If you know that enableModel1Not2 is always true in this case then you can set:
y1 = conditionalModel1.k 
however then this warning is generated (using Dymola 2024x Beta3 with Advanced.Modelica.CheckConditionallyEnabled = true):
For path conditionalModel1.k

_Using conditionally enabled variable conditionalModel1 outside of connect.
In class ExtractVariableFromConditionalModelEx.
Used in component y1.
Modelica Text: line 7_

conditionalModel1.k and conditionalModel1.k are not connectors so connect statements cannot be used. 

Obviously in this case k is the output of the model y, so it would be possible to use connect statements with  conditionalModel1.y and conditionalModel1.y to y1.  However if the conditionalModel1 and conditionalModel2 models were encrypted and users would like to display the information about a component at the top level of the model (say using the UserInteraction library components) it is not be possible without generating warnings (?).

Is there some way that I'm not aware of to get around this issue? 
Should some language change be done to support this case?

beutlich, 2024-01-14:
@HansOlsson I do not think that this is possible with current MoLang.

HansOlsson, 2024-01-15:
> @HansOlsson I do not think that this is possible with current MoLang.

Me neither, and I think it should be moved to the ModelicaSpecification issue tracking.