modelica/ModelicaStandardLibrary #4242: Impossible to choose initial excitation current on ElectricalExcited Synchronous machines?

ceraolo, 2023-12-18:
I have two models that should behave exactly the same.
The two models are as follows (please remove ".txt" before using them):
[ZeroIniIexc.mo.txt](https://github.com/modelica/ModelicaStandardLibrary/files/13704357/ZeroIniIexc.mo.txt)
[NonZeroIniExc.mo.txt](https://github.com/modelica/ModelicaStandardLibrary/files/13704358/NonZeroIniExc.mo.txt)

I get important differences in the excitation currents (both in OM and Dymola):
[image]

This is probably due to the fact that I did not specify the initial excitation current, and therefore the two simulations choose different values.
Unfortunately, it seems that there is no way to specify this initial current.

If I am not mistaken and this is true, this should be fixed for a couple of reasons
- one would have the change of starting at a given initial excitationcurrent to avoid long inisitalization times
- discrepancies between results as those reported here are avoided.

AHaumer, 2024-01-14:
There's already present the line
`  output SI.Current ie=pin_ep.i "Excitation current";`
@ceraolo why do you have to use `lesigma(i(start=0.4)`?
Like shown in the examples (e.g. Modelica.Electrical.Machines.Examples.SynchronousMachines.SMEE_DOL) one could use either an initial equation `smee.ie=0.4` or a modifier `smee(ie(start=0.4))`
Ok we'll add ie(start=0) to Modelica.Electrical.BasicMachines.SynchronousMachines.SM_ElectricalExcited - and not to forget Modelica.Magnetic.FundamentalWave.BasicMachines.SynchronousMachines.SM_ElectricalExcited.

ceraolo, 2024-01-14:
> There's already present the line ` output SI.Current ie=pin_ep.i "Excitation current";` @ceraolo why do you have to use `lesigma(i(start=0.4)`?

Well, What I did was to have a look at the machine's diagram and there I saw that the excitation inductor was Lesigma. So I just thought to set the initial current for that inductor. I did not realise that I could use ie as well.

> Ok we'll add ie(start=0) to Modelica.Electrical.BasicMachines.SynchronousMachines.SM_ElectricalExcited - and not to forget Modelica.Magnetic.FundamentalWave.BasicMachines.SynchronousMachines.SM_ElectricalExcited.

Thank you

 @christiankral what's your opinion: Should we adapt the examples changing the initial equation to the use of the new start-value?