modelica/ModelicaStandardLibrary #4244: Result Compare fail caused by atan2 in Modelica Standard Library

TongYuan-MC, 2024-01-06:
## This phenomenon has a negative impact on the following situations
1. Software benchmarking with other Modelica tool venders
2. Tool certification
3. Simulation in Critical Control System

## Example for illustration

`atan2(x, y)`  will change significantly when `x` and `y` change slightly near 0 (same issue for `atan3`). See blow:

[image]

The following simple Modelica model which also show this:

``` modelica
model Atan2Issue "z1 or z2 may differ sigificantly around pi while abs(x - y) is small"
  Real x(start = 0);
  Real y = sin(time);
  Real absxy = abs(x - y);
  Real z1, z2;
  annotation(experiment(Interval=0.0001,StartTime=0,StopTime=8));
equation
  der(x) = cos(time);  // x and y are equal in mathematical sense
  z1 = atan2(y, x);
  z2 = atan2(x, y);
end Atan2Issue;
``` 

We can see for Atan2Issue:
1. Different Modelica tools with same numerical setting has non-negligible error. 
5. Same Modelica tool with same numerical setting except tolerance also has non-negligible error. 

### Modelica Tool:
1. **MWORKS**: MWORKS.Sysplorer 2024a
2. **Dymola**: Dymola 2024x trival version
3. **OpenModelica**: OpenModelica 1.22.1

<details>
<summary> Simulation Result of Atan2Issue by MWORKS, Dymola and OpenModelica </summary>

### Atan2Issue 

`z1` and `z2` should be $\pm \frac{\pi}{4}$ in mathematical sense. However `x` was got by numerical method such as Dassl,  small differences in x and y near 0 are overamplified by `atan2`. As a result, `z1` and `z2` are unpredictable near integer multiples of $\pi$.

####  MWORKS with Tolerance `1e-6`(default) and `1e-7`
[image]

####  Dymola with Tolerance `1e-6`(default) and `1e-7`
[image]

####  OpenModelica with Tolerance `1e-6`(default) and `1e-7`
[image]

####  MWORKS, Dymola and OpenModelica with Tolerance `1e-6`(default)
[image]

####  MWORKS, Dymola and OpenModelica with Tolerance `1e-7`
[image]

</details>

**There are many models in Modelica Standard Library using atan2 and atan3**, This leads to problems when doing result comparing.  

The following two example models in **Modelica Standard Library 4.0** have same issue as model `Atan2Issue`
<details>
<summary> Simulation Result of  <b>SMEE_DOL</b> by  MWORKS, Dymola and OpenModelica </summary>

###  Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.SynchronousMachines.SMEE_DOL (Modelica Standard Library 4.0)

####  MWORKS with Tolerance `1e-6`(default) and `1e-7`
[image]

#### Dymola with Tolerance `1e-6`(default) and `1e-7`
[image]

#### OpenModelica with Tolerance `1e-6`(default) and `1e-7`
[image]

####  MWORKS, Dymola and OpenModelica with Tolerance `1e-6`(default)
[image]

#### MWORKS, Dymola and OpenModelica with Tolerance `1e-7`
[image]

</details>

<details>
<summary> Simulation Result of  <b>SoftStarter</b> by Dymola 2024x and OpenModelica 1.22.1 </summary>

###  Modelica.Electrical.PowerConverters.Examples.ACAC.SoftStarter (Modelica Standard Library 4.0)

####  MWORKS with Tolerance `1e-6`(default) and `1e-7`
[image]

#### Dymola with Tolerance `1e-6`(default) and `1e-7`
[image]

#### OpenModelica with Tolerance `1e-6`(default) and `1e-7`
[image]

####  MWORKS, Dymola and OpenModelica with Tolerance `1e-6`(default)
[image]

####  MWORKS, Dymola and OpenModelica with Tolerance `1e-7`
[image]

</details>

Could you give us some advice on dealing with such models?
1. Is there an alternative to the `atan2` and `atan3` function when Modeling?
4. Will Modelica Standard Library updating to avoid this phenomenon?
5. Is there any mechanism at the language level such as `HideResult` to avoid this phenomenon?

tobolar, 2024-01-08:
FYI: Dymola 2024x, 2023-10-06
```
Modelica.Math.atan2(0, 1e-14);
 = 0.0
Modelica.Math.atan2(1e-14, 1e-14);
 = 0.7853981633974483
Modelica.Math.atan2(-1e-14, 1e-14);
 = -0.7853981633974483
Modelica.Math.atan2(-1e-14, -1);
 = -3.1415926535897833
Modelica.Math.atan2(1e-14, -1);
 = 3.1415926535897833
```

HansOlsson, 2024-01-08:
> FYI: Dymola 2024x, 2023-10-06
>
> ```
> Modelica.Math.atan2(0, 1e-14);
>  = 0.0
> Modelica.Math.atan2(1e-14, 1e-14);
>  = 0.7853981633974483
> Modelica.Math.atan2(-1e-14, 1e-14);
>  = -0.7853981633974483 [...]

That is exactly as described in the C/C++-standard, when implementing IEC 60559 (normative appendix F). I thus cannot see that atan2 is a problem.

We would need more information which specific models cause problems - it could be that the result of atan2 should be stored, but not compared for regression testing. (Using HideResult is not a good idea.)

TongYuan-MC, 2024-01-09:
We are not saying there's something wrong with `atan2`.

Our ideal is numerical algorithms always have errors, for model `x(start = 0); der(x) = cos(time)`,  `x` won't be exact `sin(time)`, so `atan2(x, sin(time))` will not exact $\pm \frac{\pi}{4}$, and unpredictable when time is near an integer multiples of $\pi$.

When modifying certain strategies of numerical algorithms, we always get result comparison failures.

Please click the simulation result in origin comment, see blow:
[image]

henrikt-ma, 2024-01-09:
I believe this (issue or not) should be discussed within https://github.com/modelica-tools/csv-compare instead.

beutlich, 2024-01-09:
Or should the corresponding reference signals of SoftStarter and SMEE_DOL be removed?

henrikt-ma, 2024-01-10:
> Or should the corresponding reference signals of SoftStarter and SMEE_DOL be removed?

I haven't checked all of the signals mentioned above, but if we take `SoftStarter`'s `imc.rotorCage.arg_V.m` for example, it is not among the comparison signals: https://github.com/modelica/MAP-LIB_ReferenceResults/blob/v4.0.0/Modelica/Electrical/PowerConverters/Examples/ACAC/SoftStarter/comparisonSignals.txt

AHaumer, 2024-01-10:
I disagree in removing comparison signals, this makes difficult comparing versions.
We should shift the discussion to a "general" atan2-discussion, not related to special models.