modelica/ModelicaStandardLibrary #4245: Connect(p, p) in Modelica.Electrical.Analog.Examples.Utilities.SwitchedCapacitor?

TongYuan-MC, 2024-01-06:
What is the purpose of connect(p, p)？Whether it is a valid connect-equation?
[image]

AHaumer, 2024-01-06:
I suppose that is a bug. 
If you release the mouse button during a connection too early, you get such a connection with zero length.
To me it seems that it doesn't affect the results, but we should remove it.

HansOlsson, 2024-01-08:
> I suppose that is a bug. If you release the mouse button during a connection too early, you get such a connection with zero length. To me it seems that it doesn't affect the results, but we should remove it.

Agreed. Specifically it doesn't impact the result if there are other connection(s) to p, and two lines above we see another connection to p: `connect(IdealCommutingSwitch1.n1, p)`.

closed 2024-01-14 by AHaumer