modelica/ModelicaStandardLibrary #4270: Issues with tables

AHaumer, 2024-01-16:
@beutlich pls. try to simulate `Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_Conveyor` or `Modelica.Electrical.Machines.Examples.InductionMachines.IMC_Conveyor`:
Using Dympla 2024x on Win11, I get the following error:
Compilation of [Modelica.Electrical.Machines.Examples.InductionMachines.IMC_Conveyor](modelica://Modelica.Electrical.Machines.Examples.InductionMachines.IMC_Conveyor) failed.

Compiler message:

Compiling and linking the model (Visual C++). 

dsmodel.c
dsmodel.c(232): error C2065: "ModelicaStandardTables_CombiTimeTable_init3": nichtdeklarierter Bezeichner
dsmodel.c(232): error C2064: Ausdruck ergibt keine Funktion, die 0 Argumente bernimmt

Error generating Dymosim.

Failed

AHaumer, 2024-01-16:
ok then it's time to resolve this issue. I mean not specificly this one now, but in general avoiding to break regression testing.

beutlich, 2024-01-16:
Certainly, it will be resolved for MSLv4.1.0-alpha.1.

m-kessler, 2024-02-16:
For anyone stumbling across this: It looks like PR #4250 has not been merged into alpha 1, also not into beta 1.

However, the binaries from [beutlich/ModelicaStandardLibrary](https://github.com/modelica/ModelicaStandardLibrary/commit/41c4ef820adb7bb8dddae7c473e8edeb8a2613ca) work.

Dymola users have to set `Advanced.Translation.PlaceDymolaSourceFirst=2`.

AHaumer, 2024-06-11:
Could we please finally resolve this issue?
It is boring to work on other issues and not being able to simulate!
I'm not willing to fiddle around instead of doing meaningful work!
What's your opinion @casella ?

casella, 2024-06-18:
@AHaumer have you tried using MSL 4.1.0 from the HEAD of the maint/4.1.0 branch? That should work fine.

beutlich, 2024-09-09:
> Will be resolved by #4250.

Closing after #4250 got merged.

closed 2024-09-09 by beutlich