modelica/ModelicaStandardLibrary #4293: Wrong results for Modelica.Utilities.Files.loadResource()

RobertReiser, 2024-01-31:
The function `Modelica.Utilities.Files.loadResource()` returns incorrect results.

Example:
`Modelica.Utilities.Files.loadResource("modelica://MyLibrary/");`
`= "D:/Work/modelica:/MyLibrary/"`

The correct result is:
`= "D:/Libraries/MyLibrary/"`

Tested with Dymola 2023x and Dymola 2024x on Windows and Linux.

HansOlsson, 2024-01-31:
In order to test the main branch with a released Dymola you need:
1. An updated ModelicaServices 4.1.0 for Dymola (*) 
[ModelicaServices 4.1.0.zip](https://github.com/modelica/ModelicaStandardLibrary/files/14109155/ModelicaServices.4.1.0.zip) Either placed next Modelica and ModelicaServices, or just somewhere on the ModelicaPath (including in %Dymola%/Modelica/Libraries).
2. Set the flag: Advanced.Translation.PlaceDymolaSourceFirst=2;

*: Alternatively you can just manually load the ModelicaServices from the Dymola distribution after unloading the other one, but that can be a bit messy.

RobertReiser, 2024-01-31:
It works with the updated ModelicaServices library, thanks!

closed 2024-01-31 by RobertReiser