modelica/ModelicaStandardLibrary #4295: ContinuousClock has insufficient smoothness for many models in ModelicaTest.Tables

henrikt-ma, 2024-01-31:
This affects many models, but let's talk about `Modelica-4.1.0/regression/Tables/CombiTable1Ds/Test3` to make things concrete.

After `Modelica.Blocks.Sources.ContinuousClock` was modified in https://github.com/modelica/ModelicaStandardLibrary/pull/4150, the model no longer builds in System Modeler, since the input to the table isn't consider sufficiently differentiable to compute the second order derivative of the output.

Before the change to `ContinuousClock`, the input to the block was a piecewise smooth signal, and by not switching between the `if`-equation during continuous-time integration, there was no lack of smoothness.

Would it make sense to replace the `ContinuousClock` source blocks with something more differentiable?

beutlich, 2024-02-15:
> Would it make sense to replace the `ContinuousClock` source blocks with something more differentiable?

Would `Modelica.Blocks.Sources.RealExpression(y=time)` be of rescue?

henrikt-ma, 2024-02-16:
> > Would it make sense to replace the `ContinuousClock` source blocks with something more differentiable?
> 
> Would `Modelica.Blocks.Sources.RealExpression(y=time)` be of rescue?

It would save System Modeler, but let's not worry about that; we'll fix that on the tool side, as decided (but still not noted) in the last MAP-Lib release preparation meeting.

henrikt-ma, 2024-02-16:
Closing, as it was concluded that this is a tool issue.  I'll open another issue regarding the lack of excitation in the second order derivative.

closed 2024-02-16 by henrikt-ma