modelica/ModelicaStandardLibrary #4305: MSL v4.1.0-beta.1 feedback: Failing models

beutlich, 2024-02-10:
When running the [ReleaseChecks](https://github.com/modelica-tools/msl-release) in Dymola 2024x, some example/test models failed.

- [ ] `Modelica.Fluid.Examples.TraceSubstances.RoomCO2WithControls does not terminate (also when playing with tolerance settings), see also https://www.ltx.de/download/MA/Compare_MSL_v4.1.0/Testruns/Dymola/Modelica/_report/Modelica.Fluid.Examples.TraceSubstances.RoomCO2WithControls/testcase_report.html
- [ ] `ModelicaTest.Blocks.Exponentiation` raises a model error
- [ ] `ModelicaTest.Math.TestBooleanFunctions` raises an assertion

casella, 2024-02-13:
Modelica.Fluid.Examples.TraceSubstances.RoomCO2WithControls has issues also on other tools (e.g. [OpenModelica](https://libraries.openmodelica.org/branches/master/Modelica_trunk/files/Modelica_trunk_Modelica.Fluid.Examples.TraceSubstances.RoomCO2WithControls.sim)) due to chattering.

ModelicaTest.Blocks.Exponentiation is probably bogus (raising 0^0), the component that does that should removed or fixed

ModelicaTest.Math.TestBooleanFunctions is probably a Dymola bug because the model was changed, Dymola will will be fixed for the future and we can have a workaround for 4.1.0 only.

casella, 2024-02-13:
@HansOlsson will open tickets about the other two issue.

beutlich, 2024-02-16:
Closing as there are individual issues for the three cases.

closed 2024-02-16 by beutlich