modelica/ModelicaStandardLibrary #4306: MSL v4.1.0-beta.1 feedback: Immoderate size of simulation result data for LightningSegmentedTransmissionLine

beutlich, 2024-02-10:
Modelica.Electrical.Analog.Examples.Lines.LightningSegmentedTransmissionLine generates GB of result data and makes this example model hardly usuable. Also the Dymola commands readTrajectory/readTrajectorySize for saving as CSV file are likely to fail (due to limited RAM) or take an immoderate long time.

```
Name                                    Size            Date                    Attr
LightningSegmentedTransmissionLine.mat	8.889.530.094	10.02.2024 11:55	-a--
```

```
$ matdump LightningSegmentedTransmissionLine.mat -f whos
Name                       Size           Bytes          Class

Aclass                     4x11                 88  mxCHAR_CLASS
name                      50x8955           895500  mxCHAR_CLASS
description               95x8955          1701450  mxCHAR_CLASS
dataInfo                   4x8955           286560  mxDOUBLE_CLASS
data_1                  3286x2               52576  mxDOUBLE_CLASS
data_2                  2222x500002     8888035552  mxDOUBLE_CLASS
```

It really would be helpful to spend some HideResult annotations on the line1 and line2 variables.

See also #3349.

casella, 2024-02-13:
@AHaumer will look into that

AHaumer, 2024-02-13:
@Harisankar-Allimangalath @arunkumar-narasimhan as soon as  #4315 has been merged into master, please back-port to 4.1.0

commit 839ab4de (2024-02-27): resolves #4306
commit ace6673c (2024-02-27): Merge pull request #4315 from AHaumer/issue4306

resolves #4306
closed 2024-02-27 by AHaumer