modelica/ModelicaStandardLibrary #4309: MSL v4.1.0-beta.1 feedback: Inconsistent experiment annotations in Modelica.Electrical.PowerConverters.Examples.DCDC.ExampleTemplates

beutlich, 2024-02-10:
There are 3 example templates in Modelica.Electrical.PowerConverters.Examples.DCDC.ExampleTemplates, but only one of them (ChopperBuckBoost) has an experiment annoation.

https://github.com/modelica/ModelicaStandardLibrary/blob/da5c5de71f574b33a4f4631e09ba79682755f6d0/Modelica/Electrical/PowerConverters/Examples/DCDC/ExampleTemplates/ChopperBuckBoost.mo#L112-L115

I'd recommend to remove that experiment annotation on the partial models such that none of the three templates contains an experiment annotation.

casella, 2024-02-13:
@Harisankar-Allimangalath @arunkumar-narasimhan, can you pleas port @AHaumer's commit to the maintenance branch and close this ticket. Thanks!

beutlich, 2024-02-21:
Resolved in both master and maint/4.1.0.

commit 43e81b91 (2024-02-13): resolves #4309 (#4310)
commit 75e2380a (2024-02-15): resolves #4309 (#4310)
closed 2024-02-21 by beutlich