modelica/ModelicaStandardLibrary #4312: Modelica.Fluid.Examples.TraceSubstances.RoomCO2WithControls is problematic because of chattering

casella, 2024-02-13:
As reported in #4305, Modelica.Fluid.Examples.TraceSubstances.RoomCO2WithControls is problematic: Dymola 2024x [fails with it](https://www.ltx.de/download/MA/Compare_MSL_v4.1.0/Testruns/Dymola/Modelica/_report/Modelica.Fluid.Examples.TraceSubstances.RoomCO2WithControls/testcase_report.html) and so does [OpenModelica](https://libraries.openmodelica.org/branches/master/Modelica_trunk/files/Modelica_trunk_Modelica.Fluid.Examples.TraceSubstances.RoomCO2WithControls.sim). Apparently there are some reasons to generate chattering.

This should be investigated before the 4.1.0 release.

maltelenz, 2024-10-21:
We also see chattering around the same time in System Modeler.

casella, 2024-10-24:
I'll have a closer look ASAP

MatthiasBSchaefer, 2025-01-21:
This is the commit, causing the chattering according to my tests in Dymola:
https://github.com/modelica/ModelicaStandardLibrary/commit/930456a53476f9973ada6f44dbf7dda595ee4cba

HansOlsson, 2025-01-21:
I tried to debug it and preliminarily it seems that the linear part makes sense, but needed to be applied in additional cases.

Specifically the code has:
```
      if  xi1 < x0 or xi2>x1 then
        // The limits don't make sense, just use linear interpolation
        y := (y1-y0)*(x-x0)/(x1-x0) + y0;
      else
```
And what happens here is that `xi1` and `xi2` are practically identical, but `x0` and `x1` are different.
Adding `or abs(xi2-xi1)<100*Modelica.Constants.eps ` fixes this specific case, but it should be investigated a bit more.

HansOlsson, 2025-01-21:
I tried to apply the above and run all the Fluid-tests (without looking at the results) (on maint/4.1.x since I needed MSL-binaries).

In Dymola 2025x (updated) with `Advanced.Translation.PlaceDymolaSourceFirst = 2;` all fluid examples ran.
I'll make a proper PR of this.

casella, 2025-02-09:
It seems that @HansOlsson managed to fix the remaining issues with regfun3 in PR #4522. When that is merged on master we can check if there are any issues popping up with other models and if we have no evidence of that we can back-port it to 4.1.0.

casella, 2025-02-09:
As already noted in #3758, some of these issues are popping up because we are using flow models that are meant to handle static head in the trivial cases where the pipe is exactly horizontal. As I understand, this makes the numerical behaviour of regfun3 tricky. As suggested by @sielemann in [#3758](https://github.com/modelica/ModelicaStandardLibrary/issues/3758#issuecomment-1514233700), we should probably fix Modelica.Fluid.Pipes.BaseClasses.FlowModels.PartialGenericPipeFlow so that it completely skips the part that accounts for static head if max(abs(dheights)) < eps.

Given the quite complex structure of the pipe model, this is not completely trivial, so I'd keep this as an improvement for MSL 4.2.0.

casella, 2025-02-09:
Reopen until back-ported to maint/4.1.x

beutlich, 2025-02-12:
> please back-port [#4522](https://github.com/modelica/ModelicaStandardLibrary/pull/4522) to maint/v4.1.x.

Done by #4533.

casella, 2025-02-24:
As noted [here](https://github.com/modelica/ModelicaStandardLibrary/pull/4522#issuecomment-2679691759) when merging #4522 on master there was only one positive effect, namely RoomCO2WithControl starting to run properly, and no negative side effects, so I'm confident this will work out.

As soon as LTX servers have run again we should see all green [here](https://www.ltx.de/download/MA/Compare_MSL_v4.1.0/Compare/Modelica/TestcaseReports/Modelica.Fluid.Examples.TraceSubstances.RoomCO2WithControls/testcase_report.html), then we can close this ticket as well.

casella, 2025-02-25:
All is set now.

commit f6a484ca (2025-02-09): Fix #4312 (#4522)

* Avoid chattering in Modelica.Fluid.Examples.TraceSubstances.RoomCO2WithControls

* Better way of avoiding ill-posed problem.
commit 30a85923 (2025-02-24): Fix #4312 (#4533)

* Avoid chattering in Modelica.Fluid.Examples.TraceSubstances.RoomCO2WithControls

* Better way of avoiding ill-posed problem.
closed 2025-02-25 by casella