modelica/ModelicaStandardLibrary #4316: MSL v4.1.0-beta.1 feedback: broken Dymola plot scripts

m-kessler, 2024-02-14:
There are 3 plot scripts in MSL 4.1.0 beta 1 with errors:

1.  _plotResults_ in `Modelica.Fluid.Examples.HeatingSystem`
    `Error: Could not plot variable 'pipe.heatPort[1].Q_flow'`

2. _plot level and port.m_flow_ in `Modelica.Fluid.Examples.Tanks.ThreeTanks`
    `Error: Could not plot variable 'tank3.ports[2].m_flow'`

3. _plot level and port.m_flow_ in `Modelica.Fluid.Examples.Tanks.TanksWithOverflow`
    `Error: Could not plot variable 'tank3.ports[2].m_flow'`

Tested with Dymola Version 2024x Refresh 1 Alpha.

HansOlsson, 2024-02-14:
The model errors seem to be:
1. heatPort is disabled for the model, and thus there is no heatPort
2. There is only one array element for tank3.ports[:] - not two.
3. There is no tank3

beutlich, 2024-02-15:
@m-kessler I wonder how you found out? And I also wonder if there is some similar check for the figure annotation.

HansOlsson, 2024-02-15:
I checked Modelica 3.2.2 and it also lacked those variables, and at least the first script had the same issue (those script say they were generated in 2006). It seems this is a very old problem, and since there is nothing linking the model and the script I can see why it has been missed.

The figure annotation will be a better replacement.

maltelenz, 2024-02-15:
Could the existing plot scripts for Dymola be used to generate figure annotations, to get the project of adding figures on the way? (For after 4.1.0 of course.)

m-kessler, 2024-02-15:
> @m-kessler I wonder how you found out? And I also wonder if there is some similar check for the figure annotation.

We have an in-house tool that checks all our libraries at Dassault Systemes. It simulates all examples, executes the dymola-commands in the model annotation and checks if there is an error.

beutlich, 2024-02-15:
@m-kessler Thanks for the feedback. I was wondering if there could be some CI check to check both command scripts and figure annotations for invalid references.

Ayway, would it be possible for you to file a PR to fix these three plot commands. Thanks again.

beutlich, 2024-02-15:
@Harisankar-Allimangalath Please backport to maint/4.1.0.

HansOlsson, 2024-02-15:
> @Harisankar-Allimangalath Please backport to maint/4.1.0.

I did it in #4322

Harisankar-Allimangalath, 2024-02-15:
@HansOlsson I have added @maltelenz and @m-kessler  as reviewers for the #PR4322 .  Thanks

maltelenz, 2024-02-15:
FYI @Harisankar-Allimangalath if you type #4322 instead of PR4322 github automatically creates helpful cross-links that help navigation.

PR #4321 (2024-02-15): Fix plot scripts
Closes #4316
I believe these make sense. I don't know exactly what they intended to show, but:
- wall and heater have somewhat similar heat-flows showing that you heat and it is lost in the wall later on
- in the other two cases there were already enough variables plotted in that diagram, so I don't think we need the extra one
commit ae3e4885 (2024-02-15): Update plot scripts to work.
Closes #4316
PR #4322 (2024-02-15): Fix plot scripts
Backport of #4321
Closes #4316
closed 2024-02-15 by beutlich