modelica/ModelicaStandardLibrary #4325: ContinuousClock is not suitable for testing derivatives in ModelicaTest.Tables

henrikt-ma, 2024-02-16:
When testing derivatives of table outputs, the use of `ContinuousClock` as table input is a poor choice due to its very poor excitation in the derivatives.  This is especially the case when testing second order derivatives, as the second order derivative of the source signal is zero.

In detail, my concern is that if we describe the table mapping as
```
y = f(u)
```
then the general chain rule form of the derivatives would be this:
```
y' = f_der(u, u') = f'(u) * u'
y'' = f_der2(u, u', u'') = f''(u) * u'^2 + f'(u) * u''
```
If there would be an error in the `f'(u)` factor of the `f_der2` implementation, it would go unnoticed because of the multiplication with `u''` which is zero in this case.

With more knowledge about `f_der2` is implemented, one could argue that the `f'(u)` part is probably correct, but it seems unnecessary to have test models relying on such inside information.

beutlich, 2024-02-18:
I am not sure what should be done here.

henrikt-ma, 2024-02-26:
I'd suggest changing the source block to something like `Cosine` or with suitable choice of `offset`, `amplitude`, and `phase`.  This would give variation in both the input itself and its first and second derivatives.  Of course, this would require that reference results are re-generated.

HansOlsson, 2024-02-26:
> I'd suggest changing the source block to something like `Cosine` or with suitable choice of `offset`, `amplitude`, and `phase`. This would give variation in both the input itself and its first and second derivatives. Of course, this would require that reference results are re-generated.

I think that Sine-block be more similar to the current test; it starts almost as a time - but then it becomes non-linear. 
However, both of them require special care for the amplitude and offset to ensure that the entire table is tested.

Having a Sine-block plus Clock might be simpler to ensure that there's a 2nd derivative and it increases without bound.