modelica/ModelicaStandardLibrary #4329: MSL v4.1.0-beta.1 feedback: Example name inconsistency

beutlich, 2024-02-21:
This is the topic split of #4304 where I wondered why some of the new example models are called _Demo_Something and others _Demonstrate_Something. One idea was to get rid of this Demo/Demonstrate suffix.

AHaumer, 2025-02-11:
Shift to next release sinve this is cosmetics

beutlich, 2025-02-11:
No, renaming example model is not just cosmetics as it effects the regression testing. Not having the proper names now will certainly lead to unexpected regression evaluations later (missing result -> new result).

maltelenz, 2025-02-12:
> missing result -> new result

If a model is renamed, the result files should just be renamed as well, not generated from scratch.
This step needs to be taken regardless of if it's done now or for 4.2.0.

I agree with @AHaumer, we have to stop doing things for 4.1.0 to ever get it released.

tobolar, 2025-02-12:
I usually try to omit "Demo" in model names since each example is principally *demo*nstrating some model's feature, right? So to me it is generally a bad practice which should be omitted if possible in the future. (And yes, I myself sometimes indeed use "demo" for lack of better ideas.)