modelica/ModelicaStandardLibrary #4334: MSL 4.1.0 Regressions - Fluid.Dissipation

GallLeo, 2024-02-27:
The following models fail in result comparison.
Tested revision: f9bddf86 (2024-02-16)

Changed models, need reference update after library officer check:
- [ ] ModelicaTest.Fluid.Dissipation.TestCases.PressureLoss.SeveralTestCases
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.Channel.kc_evenGapLaminar_KC
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.Channel.kc_evenGapOverall
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.Channel.kc_evenGapOverall_KC
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.Channel.kc_evenGapTurbulent
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.General.kc_approxForcedConvection
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.General.kc_approxForcedConvection_KC
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.HelicalPipe.kc_laminar
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.HelicalPipe.kc_laminar_KC
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.HelicalPipe.kc_overall
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.HelicalPipe.kc_overall_KC
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.HelicalPipe.kc_turbulent
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.HelicalPipe.kc_turbulent_KC
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.Plate.kc_laminar
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.Plate.kc_laminar_KC
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.Plate.kc_overall
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.Plate.kc_overall_KC
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.Plate.kc_turbulent
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.Plate.kc_turbulent_KC
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.StraightPipe.kc_laminar
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.StraightPipe.kc_laminar_KC
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.StraightPipe.kc_overall
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.StraightPipe.kc_overall_KC
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.StraightPipe.kc_turbulent
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.StraightPipe.kc_turbulent_KC
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.StraightPipe.kc_twoPhaseOverall_KC
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.PressureLoss.General.dp_idealGas_DPMFLOW
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.PressureLoss.General.dp_nominalDensityViscosity_DPMFLOW
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.PressureLoss.General.dp_nominalPressureLossLawDensity_DPMFLOW
      - Reason: 
      - Updated reference files?

Changed output settings or Dymola flags (reference update after library officer check):
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.HeatTransfer.General.kc_approxForcedConvection_KC	
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.Dissipation.Verifications.PressureLoss.Valve.dp_severalGeometryOverall_DPMFLOW	
      - Reason: 
      - Updated reference files?

- [ ] Release notes check: Are all classes mentioned which could lead to result changes in user models?	  

---
_Useful Links_

_Current comparison report:
https://www.ltx.de/download/MA/Compare_MSL_v4.1.0/comparison_report_overview.html
-> Reference result test -> Comparison_

_Comparison signal definitions:
https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Resources/Reference/Modelica
https://github.com/modelica/ModelicaStandardLibrary/tree/master/ModelicaTest/Resources/Reference/ModelicaTest_

_Reference results:
https://github.com/modelica/MAP-LIB_ReferenceResults_

arunkumar-narasimhan, 2024-07-05:
@GallLeo, @casella, all these failed reference comparisons now pass per 2024-July-04, no need for any action.

casella, 2024-07-05:
Indeed, all green. Very good!

@GallLeo any clue why you got regressions when you opened this ticket?

closed 2024-07-05 by casella