modelica/ModelicaStandardLibrary #4335: MSL 4.1.0 Regressions - Fluid

GallLeo, 2024-02-27:
The following models fail in result comparison.
Tested revision: f9bddf86 (2024-02-16)

Changed models, need reference update after library officer check:

- [ ] Modelica.Fluid.Examples.BranchingDynamicPipes	
      - Reason: 
      - Updated reference files?
- [ ] Modelica.Fluid.Examples.Explanatory.MomentumBalanceFittings	
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.TestComponents.Fittings.TestMultiPort
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.TestComponents.Fittings.TestJunctionTraceSubstances
	  - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.TestComponents.Fittings.TestSharpEdgedOrifice
      - Reason: 
      - Updated reference files?	  
- [ ] ModelicaTest.Fluid.TestComponents.Pipes.DynamicPipesWithTraceSubstances
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.TestComponents.Sensors.TestPressure
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.TestComponents.Sources.TestSources
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.TestComponents.Valves.TestCheckValve
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.TestComponents.Valves.TestValveCompressible
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.TestComponents.Valves.TestValveDiscrete
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.TestComponents.Valves.TestValveLinear
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.TestComponents.Valves.TestValvesCompressibleReverse
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.TestComponents.Valves.TestValvesIncompressibleReverse
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.TestComponents.Valves.TestValveVaporizing
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.TestExamplesVariants.BranchingDynamicPipes_MomentumSteadyState
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Fluid.TestUtilities.Test02RegFun3
      - Reason: 
      - Updated reference files?
	  
Not classified yet, needs library officer check:	  
- [ ] Modelica.Fluid.Examples.AST_BatchPlant.BatchPlant_StandardWater
      - Reason: 
      - Updated reference files?
	  
- [ ] Release notes check: Are all classes mentioned which could lead to result changes in user models?	  	  

---
_Useful Links_

_Current comparison report:
https://www.ltx.de/download/MA/Compare_MSL_v4.1.0/comparison_report_overview.html
-> Reference result test -> Comparison_

_Comparison signal definitions:
https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Resources/Reference/Modelica
https://github.com/modelica/ModelicaStandardLibrary/tree/master/ModelicaTest/Resources/Reference/ModelicaTest_

_Reference results:
https://github.com/modelica/MAP-LIB_ReferenceResults_

GallLeo, 2024-10-20:
Currently simulation fails for:
ModelicaTest.Fluid.TestComponents.Vessels.TestSimpleTank

Maybe also because of tighter tolerance?
Did not investigate, yet.

`Failed condition: tank.medium.T >= 272.15 and tank.medium.T <= 403.15`

See simulation log and setup, here:
https://www.ltx.de/download/MA/Compare_MSL_v4.1.0/Testruns/Dymola/ModelicaTest/_report/ModelicaTest.Fluid.TestComponents.Vessels.TestSimpleTank/testcase_report.html

GallLeo, 2024-10-20:
ModelicaTest.Fluid.TestComponents.Fittings.TestJunctionTraceSubstances 
did not indicate " @GallLeo please generate reference file."

But, according to the analysis, I also updated the reference.
Is that OK @casella?

casella, 2024-10-20:
> ModelicaTest.Fluid.TestComponents.Fittings.TestJunctionTraceSubstances did not indicate " @GallLeo please generate reference file."
> 
> But, according to the analysis, I also updated the reference. Is that OK @casella?

Yes, we obviously overlooked that. I edited the analysis adding the request a posteriori. Thanks!

casella, 2024-10-20:
> Currently simulation fails for: ModelicaTest.Fluid.TestComponents.Vessels.TestSimpleTank

This also happens with OpenModelica, see [simulation log](https://libraries.openmodelica.org/branches/master/ModelicaTest_4.1.0-beta.om/files/ModelicaTest_4.1.0-beta.om_ModelicaTest.Fluid.TestComponents.Vessels.TestSimpleTank.sim).

> `Failed condition: tank.medium.T >= 272.15 and tank.medium.T <= 403.15`

Same for OMC
 
> See simulation log and setup, here: https://www.ltx.de/download/MA/Compare_MSL_v4.1.0/Testruns/Dymola/ModelicaTest/_report/ModelicaTest.Fluid.TestComponents.Vessels.TestSimpleTank/testcase_report.html

The link is unfortunately broken, this is what I get
[image]

casella, 2024-10-20:
We also have issues with Modelica.Fluid.Examples.TraceSubstances.RoomCO2WithControls. This has been known for a while, see #4312.

maltelenz, 2025-02-11:
I believe everything that remains is covered in other issues. Feel free to reopen if anyone disagrees.

Closing.

closed 2025-02-11 by maltelenz