modelica/ModelicaStandardLibrary #4336: MSL 4.1.0 Regressions - Magnetic.FluxTubes

GallLeo, 2024-02-27:
The following models fail in result comparison.
Tested revision: f9bddf86 (2024-02-16)

Changed models, need reference update after library officer check:
- [ ] Modelica.Magnetic.FluxTubes.Examples.MovingCoilActuator.ForceCurrentBehaviour
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Magnetic.FluxTubes.Sensors
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Magnetic.FluxTubes.Shapes.FixedShape
      - Reason: 
      - Updated reference files?

- [ ] Release notes check: Are all classes mentioned which could lead to result changes in user models?	  	  

---
_Useful Links_

_Current comparison report:
https://www.ltx.de/download/MA/Compare_MSL_v4.1.0/comparison_report_overview.html
-> Reference result test -> Comparison_

_Comparison signal definitions:
https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Resources/Reference/Modelica
https://github.com/modelica/ModelicaStandardLibrary/tree/master/ModelicaTest/Resources/Reference/ModelicaTest_

_Reference results:
https://github.com/modelica/MAP-LIB_ReferenceResults_

AHaumer, 2024-03-01:
In Modelica.Magnetic.FluxTubes.Examples.MovingCoilActuator.ForceCurrentBehaviour the variable pmActuator.coil.L_stat is just a bad choice. For numerical reasons it gets protected from a division by zero (when the current through the coil is near zero). This depends on interval (which is not defined!) and tolerance.
Question 1: Should we exclude the mentioned variable from comparisonSignals.txt?
Question 2: Should we define a meaningful interval in the experiment annotation for all 3 examples in Modelica.Magnetic.FluxTubes.Examples.MovingCoilActuator?
@christiankral @GallLeo what's your opinion?

christiankral, 2024-03-10:
> In Modelica.Magnetic.FluxTubes.Examples.MovingCoilActuator.ForceCurrentBehaviour the variable pmActuator.coil.L_stat is just a bad choice. For numerical reasons it gets protected from a division by [...]

Yes indeed. This is a good proposal.

> Question 2: Should we define a meaningful interval in the experiment annotation for all 3 examples in 
Modelica.Magnetic.FluxTubes.Examples.MovingCoilActuator? @christiankral @GallLeo what's your opinion?

This makes also sense. @AHaumer Please provide a PR

maltelenz, 2024-06-24:
Reopening since we need to backport #4418 to maint branch before closing.

casella, 2024-06-28:
As I understand, we also need the new reference results to be generated and uploaded.

@maltelenz I would close the tickets once the action have been taken on master. I asked Modelon India to take care of the back-porting, they can go through all the 4.1.0 issues that have been closed in the last two months, collect the commits and make one big PR with all the back-ports.

maltelenz, 2024-07-01:
> @maltelenz I would close the tickets once the action have been taken on master.

I disagree, then we would risk forgetting something for 4.1.0. If the issue is closed, what is supposed to remind us?

> collect the commits and make one big PR with all the back-ports.

That would make it very hard to verify that the backporting is correct, since you cannot compare to the changes in the original PRs if everything is thrown together in a very big pile.

casella, 2024-07-01:
We can keep it open until it gets backported. Note that we may have other tickets that have been closed already but not back-ported, so it's going to be messy anyway 😃

casella, 2024-07-31:
No impact on user's models, no need to mention in the release notes. Can be closed once backported to maint/4.1.x

maltelenz, 2025-02-11:
As far as I can tell, nothing remains.

Closing.

commit 22c3f1a6 (2024-06-24): should resolve #4336
commit 9fb423e5 (2024-06-24): Merge pull request #4418 from AHaumer/4336

Resolve #4336
PR #4447 (2024-08-19): Resolve #4336
Backport #4418 from master to main/4.1.x.

Closes #4336
commit 21b928ea (2024-08-19): Resolve #4336 (#4447)

* Remove ill-defined pmActuator.coil.L_stat from comparisonSignals.txt for Modelica.Magnetic.FluxTubes.Examples.MovingCoilActuator.ForceCurrentBehaviour
* Define output interval for all 3 examples Modelica.Magnetic.FluxTubes.Examples.MovingCoilActuator
closed 2025-02-11 by maltelenz