modelica/ModelicaStandardLibrary #4340: MSL 4.1.0 Regressions - Mechanics.Rotational and Mechanics.Translational

GallLeo, 2024-02-27:
The following models fail in result comparison.
Tested revision: f9bddf86 (2024-02-16)

Changed models, need reference update after library officer check:
- [ ] Modelica.Mechanics.Rotational.Examples.EddyCurrentBrake
      - Reason: 
      - Updated reference files?
- [ ] Modelica.Mechanics.Translational.Examples.EddyCurrentBrake
      - Reason: 
      - Updated reference files?

- [ ] Release notes check: All classes mentioned which could lead to result changes in user models?

tobolar, 2024-03-01:
All three points fine to me. 
@AHaumer after checking the points, you can please tick them all.

AHaumer, 2024-03-01:
@tobolar I won't tick the points, @GallLeo will do that as soon as he has created new reference results.

GallLeo, 2024-10-21:
@Esther-Devakirubai Can you please, whether the release notes mention EddyCurrentBrake?

maltelenz, 2025-02-11:
> [@Esther-Devakirubai](https://github.com/Esther-Devakirubai) Can you please, whether the release notes mention EddyCurrentBrake?

@GallLeo @Esther-Devakirubai this seems to be the remaining item on this issue.

AHaumer, 2025-02-11:
@GallLeo isn't this resolved now? If yes, pls. close

casella, 2025-02-24:
It is mentioned in DifferencesTo400.html, see #4300

casella, 2025-02-27:
ping @arunkumar-narasimhan

closed 2025-03-09 by casella