modelica/ModelicaStandardLibrary #4341: MSL 4.1.0 Regressions - MultiBody

GallLeo, 2024-02-27:
The following models fail in result comparison.
Tested revision: f9bddf86 (2024-02-16)

Changed models, need reference update after library officer check:
- [ ] Modelica.Mechanics.MultiBody.Examples.Elementary.PointGravityWithPointMasses2
      - Reason: 
      - Updated reference files?
- [ ] Modelica.Mechanics.MultiBody.Examples.Rotational3DEffects.BevelGear1D
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.MultiBody.Parts.FreeBodies
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.MultiBody.Sensors.AbsoluteSensor
      - Reason: 
      - Updated reference files?

No signals to compare, define more comparison signals:
- [ ] ModelicaTest.MultiBody.Visualizers.Arrow
      - Updated comparisonSignals.txt?
	  - Updated reference files?
- [ ] ModelicaTest.MultiBody.Visualizers.ColorMaps
      - Updated comparisonSignals.txt?
	  - Updated reference files?
- [ ] ModelicaTest.MultiBody.Visualizers.Shapes
      - Updated comparisonSignals.txt?
	  - Updated reference files?
- [ ] ModelicaTest.MultiBody.Visualizers.Torus
      - Updated comparisonSignals.txt?
	  - Updated reference files?

Change due to new Dymola version (reference update after library officer check?):
- [ ] ModelicaTest.MultiBody.Joints.Universal	
      - Reason: Numerical differences on accelerations (second derivative). Martin Otter (during MAP-LIB meeting): Please update reference
      - Updated reference files?
- [ ] ModelicaTest.MultiBody.Joints.UniversalSpherical
      - Reason: Numerical differences on accelerations (second derivative). Martin Otter (during MAP-LIB meeting): Please update reference
	  - Updated reference files?
- [ ] ModelicaTest.MultiBody.Parts.Rotor1D.GearConstraint3	
      - Reason: 
      - Updated reference files?
	  
Not classified, yet (needs library officer check):
- [ ] Modelica.Mechanics.MultiBody.Examples.Rotational3DEffects.BevelGear1D
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.MultiBody.PlanarLoopWithMove
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.MultiBody.Sensors.Distance2
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.MultiBody.Sensors.RelativeSensor
      - Reason: 
      - Updated reference files?

- [ ] Release notes check: Are all classes mentioned which could lead to result changes in user models?	  
	  
---
_Useful Links_

_Current comparison report:
https://www.ltx.de/download/MA/Compare_MSL_v4.1.0/comparison_report_overview.html
-> Reference result test -> Comparison_

_Comparison signal definitions:
https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Resources/Reference/Modelica
https://github.com/modelica/ModelicaStandardLibrary/tree/master/ModelicaTest/Resources/Reference/ModelicaTest_

_Reference results:
https://github.com/modelica/MAP-LIB_ReferenceResults_

tobolar, 2024-03-01:
Changed models, need reference update after library officer check:

- Modelica.Mechanics.MultiBody.Examples.Elementary.PointGravityWithPointMasses2
  Checking https://www.ltx.de/download/MA/Compare_MSL_v4.1.0/Compare/Modelica/20240227232653/Modelica.Mechanics.MultiBody.Examples.Elementary.PointGravityWithPointMasses2/CSVCompare/PointGravityWithPointMasses2_report.html,
  the error message is: `6 (relative error is 1,81520807176844E-16)`
  So I really wonder why csv-compare catch this.
  Anyway - this example is fine to me.

- Modelica.Mechanics.MultiBody.Examples.Rotational3DEffects.BevelGear1D
  I simulated both MSL400 and MSL4100 in Dymola2023x Refresh1, and then MSL400 and MSL4100 in Dymola2024x. Comparing results of the two MSL versions calculated in ONE of Dymola yields the same results. So the problem seems to be the comparison across the Dymola versions. I tried to increase the numerical tolerance but this alone doesn't help.
  What seems could help is to a) introduce a damping and b) increase the tolerance (to=1e-6).
  So maybe better place this under "Change due to new Dymola version" below?

- ModelicaTest.MultiBody.Parts.FreeBodies
  Checking https://www.ltx.de/download/MA/Compare_MSL_v4.1.0/Compare/ModelicaTest/20240228011806/ModelicaTest.MultiBody.Parts.FreeBodies/CSVCompare/FreeBodies_report.html,
  the error message is: `433 (relative error is 3,22845694576259E-13)`
  So I really wonder why csv-compare catch this.
  Anyway - this example is fine to me.

- ModelicaTest.MultiBody.Sensors.AbsoluteSensor
  Test model changed, see https://github.com/modelica/ModelicaStandardLibrary/pull/4148 (and esp. 82d1b167e20e0decf632a8c9a2ea58f3c92415cb)
  So the reference files shall be updated.

No signals to compare, define more comparison signals:

- ModelicaTest.MultiBody.Visualizers
  The tests are intended to test only 3d visualization of the objects. So not sure how to proceed here.

Change due to new Dymola version (reference update after library officer check?):

- ModelicaTest.MultiBody.Parts.Rotor1D.GearConstraint3
This seems to be similar to Modelica.Mechanics.MultiBody.Examples.Rotational3DEffects.BevelGear1D above.
So a) introduce a damping and b) increase the tolerance (to=1e-6).

Not classified, yet (needs library officer check):

- Modelica.Mechanics.MultiBody.Examples.Rotational3DEffects.BevelGear1D
  This item is doubled, see the same class above.

- ModelicaTest.MultiBody.PlanarLoopWithMove
  I can't simulate - maybe missing binaries? https://github.com/modelica/ModelicaStandardLibrary/pull/4250

- ModelicaTest.MultiBody.Sensors.Distance2
  There is no error reported.

- ModelicaTest.MultiBody.Sensors.RelativeSensor
  There is no error reported.

casella, 2024-06-18:
On 2024-06-18 I reviewed all issues with @tobolar and made decisions, and actions, reported in the top ticket description

GallLeo, 2024-07-02:
Modelica.Mechanics.MultiBody.Examples.Rotational3DEffects.BevelGear1D

I tried to reduce the reference CSV to 0.4 s.

It nearly works in csv-compare. Unfortunately, it reports errors for each variable, because the tube extrapolates a bit in the future.

[image]

Proposal:
- update reference file to stop at 0.4 s
- accept regression failure, for now.
- improve CSV-compare or agree on test-stop-time-annotation for future MSL development (CI)

GallLeo, 2024-10-21:
For the above mentioned chaotic models, I cut the CSV files manually and added a Markdown note to the reference folders:

BevelGear1D
https://github.com/modelica/MAP-LIB_ReferenceResults/blob/13debd9819062ea6ecc321538780f50717c11467/Modelica/Mechanics/MultiBody/Examples/Rotational3DEffects/BevelGear1D/readme.md

GearConstraint3
https://github.com/modelica/MAP-LIB_ReferenceResults/blob/13debd9819062ea6ecc321538780f50717c11467/ModelicaTest/MultiBody/Parts/Rotor1D/GearConstraint3/readme.md

maltelenz, 2025-02-11:
As far as I can tell, no items remain. If anyone disagrees, feel free to reopen.

Closing.

closed 2025-02-11 by maltelenz