modelica/ModelicaStandardLibrary #4342: MSL 4.1.0 Regressions - Utilities

GallLeo, 2024-02-27:
The following models fail in result comparison.
Tested revision: f9bddf86 (2024-02-16)

Changed models, need reference update after library officer check:
- [ ] ModelicaTest.Utilities.TestInternal
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Utilities.TestStreams
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Utilities.TestStrings
      - Reason: 
      - Updated reference files?  
- [ ] ModelicaTest.Utilities.TestWriteFile
      - Reason: 
      - Updated reference files?

No signals to compare, define more comparison signals:
- [ ] Modelica.Utilities.Examples.readRealParameterModel
      - Updated comparisonSignals.txt?
	  - Updated reference files?
- [ ] ModelicaTest.Utilities.TestExamplesCalculator
      - Updated comparisonSignals.txt?
	  - Updated reference files?
- [ ] ModelicaTest.Utilities.TestStrings
      - Updated comparisonSignals.txt?
	  - Updated reference files?

- [ ] Release notes check: Are all classes mentioned which could lead to result changes in user models?	  
	  
---
_Useful Links_

_Current comparison report:
https://www.ltx.de/download/MA/Compare_MSL_v4.1.0/comparison_report_overview.html
-> Reference result test -> Comparison_

_Comparison signal definitions:
https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Resources/Reference/Modelica
https://github.com/modelica/ModelicaStandardLibrary/tree/master/ModelicaTest/Resources/Reference/ModelicaTest_

_Reference results:
https://github.com/modelica/MAP-LIB_ReferenceResults_

beutlich, 2024-03-23:
I'd claim the reported 3 regressions are no longer reproducable and this issue can be closed.

casella, 2024-06-11:
No need to change references, no more regressions. Case closed

closed 2024-06-11 by casella