modelica/ModelicaStandardLibrary #4343: MSL 4.1.0 Regressions - Blocks.Tables

GallLeo, 2024-02-27:
The following models fail in result comparison.
Tested revision: f9bddf86 (2024-02-16)

Changed C-code/binaries, need reference update after library officer check:
- [ ] ModelicaTest.Tables.CombiTable2Ds.Test26
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Tables.CombiTable2Ds.Test27
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Tables.CombiTable2Dv.Test26
      - Reason: 
      - Updated reference files?
- [ ] ModelicaTest.Tables.CombiTable2Dv.Test27
      - Reason: 
      - Updated reference files?
	  
No signals to compare, define more comparison signals:
- [ ] ModelicaTest.Tables.CombiTable2Ds.Test15
      - Updated comparisonSignals.txt?
	  - Updated reference files?	  
- [ ] ModelicaTest.Tables.CombiTable2Ds.Test16
      - Updated comparisonSignals.txt?
	  - Updated reference files?	  
- [ ] ModelicaTest.Tables.CombiTable2Ds.Test17
      - Updated comparisonSignals.txt?
	  - Updated reference files?	  

- [ ] Release notes check: Are all classes mentioned which could lead to result changes in user models?	  
	  
---
_Useful Links_

_Current comparison report:
https://www.ltx.de/download/MA/Compare_MSL_v4.1.0/comparison_report_overview.html
-> Reference result test -> Comparison_

_Comparison signal definitions:
https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Resources/Reference/Modelica
https://github.com/modelica/ModelicaStandardLibrary/tree/master/ModelicaTest/Resources/Reference/ModelicaTest_

_Reference results:
https://github.com/modelica/MAP-LIB_ReferenceResults_

beutlich, 2024-02-27:
> Tested revision: [f9bddf8](https://github.com/modelica/ModelicaStandardLibrary/commit/f9bddf86b19027bf2223b3e5a29647a35550e5d5) (2024-02-16)

f9bddf86b19027bf2223b3e5a29647a35550e5d5 is not part of v4.1.0-beta.1. Should it be da5c5de71f574b33a4f4631e09ba79682755f6d0?

beutlich, 2024-02-27:
I am confused. I ran the result comparison of ModelicaTest.Tables of MSL4.1.0-beta.1 w.r.t. MSL v4.0.0 and get different comparison errors, see https://beutlich.github.io/MSLTEST410BETA1FAILED_Tables/

beutlich, 2024-02-27:
We had two bug fixes for CombiTable2D, which fixed the interpolation and 2-nd derivatives in case of extrapolation by constant continuation:

https://github.com/modelica/ModelicaStandardLibrary/blob/da5c5de71f574b33a4f4631e09ba79682755f6d0/Modelica/Resources/C-Sources/ModelicaStandardTables.c#L47-L56

MatthiasBSchaefer, 2024-03-19:
The regression tests of LTX cover the branch "master". We do NOT consider the branch v4.1.0-beta.1
development seems to be performed in both branches, which is maximal confusing ...

beutlich, 2024-03-19:
> development seems to be performed in both branches, which is maximal confusing ...

Hm, it indeed doubles the workload, but it should not be confusing.

beutlich, 2024-03-19:
> I am confused. I ran the result comparison of ModelicaTest.Tables of MSL4.1.0-beta.1 w.r.t. MSL v4.0.0 and get different comparison errors, see https://beutlich.github.io/MSLTEST410BETA1FAILED_Tables/

@MatthiasBSchaefer Any opinion on this one?

beutlich, 2024-03-21:
@GallLeo I have problems in opening the comparison links, e.g. https://www.ltx.de/download/MA/Compare_MSL_v4.1.0/Compare/ModelicaTest/20240316005556/ModelicaTest.Tables.CombiTable2Ds.Test26/CSVCompare/Test26_report.html

[image]

beutlich, 2024-03-21:
> The regression tests of LTX cover the branch "master".

Being called "MSL 4.1.0 Regressions" - they should no longer run from master, but from maint/4.1.0 branch.

beutlich, 2024-03-21:
> We had two bug fixes for CombiTable2D, which fixed the interpolation and 2-nd derivatives in case of extrapolation by constant continuation:

@HansOlsson Regarding ModelicaTest.Tables.CombiTable2Ds.Test26 the issue is in the very last step (at t = 1) of the second derivative no longer being -8 but 0 (as is the case for t > 1 s).

> d2_t_new.y is invalid! 2 errors have been found during validation.

[image]

After rolling d0d158068b8f195c3c45703fc5b91eb3359f39e4 of #3893 back, I can confirm that the regression issue is gone. To me the discontinuous second derivative looks strange.

beutlich, 2024-03-21:
> I am confused. I ran the result comparison of ModelicaTest.Tables of MSL4.1.0-beta.1 w.r.t. MSL v4.0.0 and get different comparison errors

Hm, not sure what I did then, but I finally can reproduce the regression report:

> The compare directory contained 208 files.
196 files were tested.
4 files failed.
Success rate is 98,0%.

beutlich, 2024-05-22:
@HansOlsson Can you please define the expected value of ModelicaTest.Tables.CombiTable2Ds.Test26.d2_t_new.y at time=1? I beleive it should be -8 (as is the case in MSL v4.0.0 and also sensible when loooking at d2_t_new.u) and not jump to zero.

HansOlsson, 2024-05-31:
> @HansOlsson Can you please define the expected value of ModelicaTest.Tables.CombiTable2Ds.Test26.d2_t_new.y at time=1? I believe it should be -8 (as is the case in MSL v4.0.0 and also sensible when loooking at d2_t_new.u) and not jump to zero.

I will investigate it further as there seems to be some problem for other cases, but I don't see that specific case as problematic:
If you simulate to 2 s, you see that the derivative jumps to zero after 1 s. Thus both -8 and 0 seem acceptable at 1 s.

HansOlsson, 2024-06-03:
The other case seems a lot more important; I was afraid that I had messed up something - but that doesn't seem like the case.

Consider the following two models:
```
package ProbTest
  model Test26Var
    extends ModelicaTest.Tables.CombiTable2Ds.Test26(t_new(extrapolation=Modelica.Blocks.Types.Extrapolation.HoldLastPoint));
    annotation (experiment(StopTime=2));
  end Test26Var;

  model tabletest3894
    Modelica.Blocks.Sources.Ramp ramp(
      height=1,
      duration=1,
      offset=1)
      annotation (Placement(transformation(extent={{-80,12},{-60,32}})));
    Modelica.Blocks.Sources.Constant
                                 const(k=-1)
      annotation (Placement(transformation(extent={{-80,-36},{-60,-16}})));
    Modelica.Blocks.Tables.CombiTable2Ds combiTable2Ds1(
      tableOnFile=false,
      table=[0,1,2; 1,2,3; 2,3,4],
      smoothness=Modelica.Blocks.Types.Smoothness.LinearSegments,
      extrapolation=Modelica.Blocks.Types.Extrapolation.HoldLastPoint)
      annotation (Placement(transformation(extent={{-14,-10},{6,10}})));
    Modelica.Blocks.Continuous.Derivative derivative(initType=Modelica.Blocks.Types.Init.SteadyState)
      annotation (Placement(transformation(extent={{32,-10},{52,10}})));
    Modelica.Blocks.Continuous.Der der1
      annotation (Placement(transformation(extent={{32,-40},{52,-20}})));
  equation 

    connect(combiTable2Ds1.u1, ramp.y) annotation (Line(points={{-16,6},{-50,6},{-50,
            22},{-59,22}},      color={0,0,127}));
    connect(combiTable2Ds1.u2,const. y) annotation (Line(points={{-16,-6},{-52,-6},
            {-52,-26},{-59,-26}}, color={0,0,127}));
    connect(combiTable2Ds1.y, derivative.u)
      annotation (Line(points={{7,0},{30,0}}, color={0,0,127}));
    connect(combiTable2Ds1.y, der1.u) annotation (Line(points={{7,0},{22,0},{22,
            -30},{30,-30}}, color={0,0,127}));
    annotation (
      Icon(coordinateSystem(preserveAspectRatio=false)),
      Diagram(coordinateSystem(preserveAspectRatio=false)),
      uses(Modelica(version="4.0.0")),
      experiment(StopTime=1, __Dymola_Algorithm="Dassl"),
      __Dymola_Commands(executeCall(
          ensureSimulated=true,
          autoRun=true) = {createPlot(
          id=1,
          position={0,0,1228,756},
          y={"ramp.y","ramp1.y"},
          range={0.0,1.2000000000000002,-2.0,20.0},
          grid=true,
          colors={{28,108,200},{238,46,47}},
          timeUnit="s"),createPlot(
          id=1,
          position={0,0,1228,756},
          y={"combiTable2Ds1.y","derivative.y","der1.y"},
          range={0.0,1.2000000000000002,-5.0,35.0},
          grid=true,
          subPlot=102,
          colors={{28,108,200},{0,140,72},{238,46,47}},
          timeUnit="s")} "plots"));
  end tabletest3894;
  annotation (uses(ModelicaTest(version="4.1.0"), Modelica(version="4.1.0")));
end ProbTest;
```
Currently tabletest3894 gives reasonable result, but Test26Var does after 1s have a constant signal with derivative -2. That is clearly non-sense, and blocking.

Manually reverting #3896 fixes Test26Var, but tabletest3894 then gives bad results.

beutlich, 2024-06-04:
> That is clearly non-sense, and blocking.

Thanks for finding. It is reverted by #4415.

HansOlsson, 2024-06-11:
To do:

- [ ] Just update reference results for ModelicaTest.Tables.CombiTable2D[sv].Test2[67]
- The other one:
  - [ ] Merge #4415 
  - [ ] Cherry pick #4415 to maint/4.1.x
  - [ ] Rebuild ModelicaStandardLibraries and promote on maint/4.1.x

casella, 2024-06-11:
@HansOlsson what about PR #4360? Should it also be merged or is it now obsolete? If it is obsolete, please close it with a short explanation for the record. Thanks!

@HansOlsson, @beutlich once #4415 is cherry-picked, who takes care of rebuilding the binaries?

beutlich, 2024-06-11:
> who takes care of rebuilding the binaries?

The library officers do.

casella, 2024-06-11:
@beutlich I'm sorry that I cannot really follow the discussion in #4360, as it is very technical and implementation-related. Could you provide a concrete use case or MWE that shows what the regression is all about?

Thanks

casella, 2024-06-11:
> > who takes care of rebuilding the binaries?
> 
> The library officers do.

OK. Two questions about that:
- do I understand correctly that the built libraries are made available directly in the Modelica\Resources\Library\ resource directory of the GIT repository?
- will they be available on both the master and maint/4.1.x branches?

Thanks!

beutlich, 2024-06-11:
> @beutlich I'm sorry that I cannot really follow the discussion in #4360, as it is very technical and implementation-related. Could you provide a concrete use case or MWE that shows what the regression is all about?

The example is all here in this issue. I am going to clarify with @HansOlsson such that tere will be a common understanding.

casella, 2024-06-18:
@beutlich I understand what is missing to close this ticket is
- cherry pick #4415 to maint/4.1.x
- rebuild the libraries and place them under Resources on maint/4.1.x (not master, that will be another story)

Do I miss something else?

beutlich, 2024-06-18:
> cherry pick

No, #4415 is already back-ported from master to maint/4.1.x via #4417. Hence, that's no more an issue.

> Do I miss something else?

I am actually concerned about #4360 which I did not like to se closed. So far, no clarification.

> rebuild the libraries

Can do. But I was waiting to see #4360 clarified first to avoid double work.

casella, 2024-06-18:
I'm not sure #4360 can be considered a blocker for the whole 4.1.0 release. This seems to me a really minor issue. If you can find an agreement with @HansOlsson that would be good, otherwise I'd keep it open for 4.2.0.

Thanks!

maltelenz, 2024-06-19:
> I understand what is missing to close this ticket is
> 
>    * cherry pick https://github.com/modelica/ModelicaStandardLibrary/pull/4415 to maint/4.1.x
>    * rebuild the libraries and place them under Resources on maint/4.1.x (not master, that will be another story)
>
> Do I miss something else?

Yes, the reference results need to be updated (once the binaries are rebuilt?), per earlier comments in this issue.

beutlich, 2024-06-19:
> This seems to me a really minor issue.

Not sure actually.

> If you can find an agreement with @HansOlsson that would be good

Well, I was hoping so.

beutlich, 2024-06-19:
> * rebuild the libraries and place them under Resources on maint/4.1.x

There now is #4426.

closed 2025-02-11 by casella