modelica/ModelicaStandardLibrary #4359: MSL 4.1.0 Reference results Modelica.Mechanics.MultiBody.Examples.Loops.Utilities.GasForce2

maltelenz, 2024-03-21:
In #4081 variables were removed from the comparison signals. Hence, the [reference results](https://github.com/modelica/MAP-LIB_ReferenceResults/tree/v4.1.0/ModelicaTest/MultiBody/Forces/EngineGasForce) need to be updated.

maltelenz, 2024-03-21:
Not a duplicate, the other issue is about new models.

I'm not satisfied with the fully regenerated baselines on your fork, maplib decided for higher ambitions this time, to only update references where we have a reason.

beutlich, 2024-03-21:
> Not a duplicate, the other issue is about new models.

It is a suplicate in the sense that https://github.com/beutlich/MAP-LIB_ReferenceResults gives you both the new and the altered results. Anyway, take it or leave it. But it does not help to complain again and again.

I actually wonder how MAP-Lib can figure out a-priori which simulation results change given a singular change on a base component, say e.g. https://github.com/modelica/ModelicaStandardLibrary/commit/e5f0e7eb5050456a8040d5a847adaab939cf866b. In my opinion you can only find out if you run a full regression. And in that case you get the altered results (as increment to the existing results). If a model is not affected, the CSV stays identical and git can handle it.

maltelenz, 2024-03-21:
I don't see opening issues to keep track of what still needs to be done before a release as complaining.

beutlich, 2024-03-21:
Hm, I have to admit I am even more confused. Are you asking for the (as previously done) manual job of running the simulation of a __tagged__ release in Dymola and providing them (as described in https://github.com/modelica/ModelicaStandardLibrary/wiki/Reference-results) or are you asking for the commit-triggered increments of simulation results provided automatically by a CI? For the former you have my answer, for the latter MAP-Lib is not yet there.

maltelenz, 2024-03-21:
We already have regression runs that run Dymola, comparing new results against the expected results from 4.0.0. We also have a collection of reported issues where these runs generate different results.

 The next step is that library officers make a decision for each case where the results differ. Either they decide that the new results are expected because of changes in the library, and the reference results can then be updated (for the specific model). Alternatively, the difference is unexpected, and something has to be fixed in the library before release.

Just updating all references misses the important step of verifying that the new results are correct. Even if one does verify that, updating references that were within tolerances for the regression run, risks that the results drift away over time.

Of course, doing this step for each commit or each PR during development in a CI fashion is preferable, but as you say, we aren't there yet. I believe this is the main thing @casella wants us to focus on once 4.1.0 is ready.

beutlich, 2024-03-21:
> Anyway, take it or leave it

Better leave it. After spending some more time on https://github.com/modelica/ModelicaStandardLibrary/issues/4343 I only noticed now that some of the simulation result is no longer reproducable. Not sure what I did then (beside the too much manual patching of unusable MSL 4.1.0-beta.1). In order not to use it I removed that fork for now. Sorry for the inconvenience.

closed 2025-01-27 by maltelenz