modelica/ModelicaStandardLibrary #4361: Wrong name of the release branch for MSL 4.1.0 (and following bug fix versions)

dietmarw, 2024-03-24:
It seems that whoever created the release branch for MSL version 4.1.0 under `maint/4.1.0` did not read the guidelines under:

https://doc.modelica.org/Modelica%204.0.0/Resources/helpDymola/Modelica_UsersGuide_ReleaseNotes.html#Modelica.UsersGuide.ReleaseNotes.VersionManagement

It needs to be `maint/4.1.x` since otherwise there will be:
 1. confusion as to what is the tag and what a branch
 2. There is no place for future bug fixes to live and releases to be cut from in the 4.1.x series.

This needs to be rectified **as soon as possible**.

AHaumer, 2024-03-24:
Thank you very much @dietmarw for your kind explanation, you're 100% right. 
I have renamed the branch from maint\4.1.0 to maint\4.1.x and hope that no complications will occur. 
However, a slightly more polite conversational tone would be highly appreciated.
Best regards,
Toni

dietmarw, 2024-03-24:
Sorry, I was just stating the facts.

closed 2024-03-24 by dietmarw