modelica/ModelicaStandardLibrary #4363: MSL 4.1.0 Reference results Modelica.Electrical.Machines.Examples.SynchronousMachines.SMPM_NoLoad

maltelenz, 2024-03-25:
After #4031 the reference signals for `Modelica.Electrical.Machines.Examples.SynchronousMachines.SMPM_NoLoad` changed, so we need new reference results.

beutlich, 2024-03-25:
> I also question why the test is not marked red in the report: https://www.ltx.de/download/MA/Compare_MSL_v4.1.0/Compare/Modelica/testrun_report.html

To answer this question. It is marked as `N/A` since none of the new signal names

```
toPolar.y[1]
toPolar.y[2]
hallSensor.y
```

is available in https://github.com/modelica/MAP-LIB_ReferenceResults/blob/2be1c7e08a39d658b1326607917a0a32749e76e1/Modelica/Electrical/Machines/Examples/SynchronousMachines/SMPM_NoLoad/SMPM_NoLoad.csv

maltelenz, 2024-03-25:
That sounds serious enough to warrant red coloring, if the reference result doesn't even contain the variables one expects to test...

beutlich, 2024-03-25:
Well, csv-compare writes a failed marker filer indeed: [compare_failed.log](https://github.com/modelica/ModelicaStandardLibrary/files/14746976/compare_failed.log).

MatthiasBSchaefer, 2024-04-10:
We are only counting failed comparison.
Something, that can not be compared is not a failed comparison. 

We can introduce a yellow instead of a gray field for this, if this makes you happy ...

maltelenz, 2024-04-10:
> We can introduce a yellow instead of a gray field for this, if this makes you happy ...

That sounds better than gray. It is a serious issue with the test or regression result that needs solving, so it would be nice if the report can clearly mark it so it stands out.

GallLeo, 2024-10-21:
Fixed via reference update
https://github.com/modelica/MAP-LIB_ReferenceResults/commit/6db6c501435f128346e7dcd56c4593aa2251c964

See also discussion in in modelica/ModelicaStandardLibrary/issues/4333

casella, 2024-11-11:
The test of Modelica.Electrical.Machines.Examples.SynchronousMachines.SMPM_NoLoad on the [LTX server](https://www.ltx.de/download/MA/Compare_MSL_v4.1.0/Compare/Modelica/testrun_report.html) now shows green.

casella, 2025-03-01:
While checking MSL 4.1.0 regressions in OMC, I noticed that Modelica.Electrical.Machines.Examples.SynchronousMachines.SMPM_NoLoad was passing the verification testing successfully in [MSL 4.0.0](https://libraries.openmodelica.org/branches/master/Modelica_4.0.0/Modelica_4.0.0.html), while it doesn't in [MSL 4.1.0](https://libraries.openmodelica.org/branches/master/Modelica_4.1.0-beta.om/Modelica_4.1.0-beta.om.html), due to a mismatch in [toPolar.y[2]](https://libraries.openmodelica.org/branches/master/Modelica_4.1.0-beta.om/files/Modelica_4.1.0-beta.om_Modelica.Electrical.Machines.Examples.SynchronousMachines.SMPM_NoLoad.diff.toPolar.y[2].html) 

[image]

It seems there is some issue around the points where the angle is passing from -pi to +pi. The results are matching otherwise.

maltelenz, 2025-03-03:
> [@MatthiasBSchaefer](https://github.com/MatthiasBSchaefer), [@GallLeo](https://github.com/GallLeo) why [these settings](https://github.com/modelica/MAP-LIB_ReferenceResults/blob/27f9fc2b30154c5e7321c2444e42e39d4c466e62/Modelica/Electrical/Machines/Examples/SynchronousMachines/SMPM_NoLoad/creation.txt#L17) were changed in [modelica/MAP-LIB_ReferenceResults@7d62e18](https://github.com/modelica/MAP-LIB_ReferenceResults/commit/7d62e180ad9ccaa8b7891a1fc3288ceee3814a4c)?
> 
> ```
> StartTime = 0.0    // set by user or program default value
> StopTime = 0.04    // set by user or program default value
> Interval = 5e-05    // set by user or program default value
> Tolerance = 1e-06    // set by user or program default value
> ```

They were not, it is just whitespace and comment changes?

MatthiasBSchaefer, 2025-03-03:
@casella 
The StopTime was already 0.04 before https://github.com/modelica/MAP-LIB_ReferenceResults/commit/7d62e180ad9ccaa8b7891a1fc3288ceee3814a4c  (despite of some whitespace changes)

The StopTime was changed in https://github.com/modelica/ModelicaStandardLibrary/commit/01c3f90f1b0050045008122c15153008b029a03b, but never used for new reference files, which is unfortunately a bit complicated in our tooling:
We need ONE single source to get the right experiment-settings (completely: tolerance, outputInterval, StopTime,...).
Currently this is the creation.txt to get the tighter tolerance for testing and such changes in stopTome are ignored.

casella, 2025-03-03:
> We need ONE single source to get the right experiment-settings (completely: tolerance, outputInterval, StopTime,...). 

I couldn't agree more. 

I actually made a proposal to address this issue with suitable annotations, see modelica/ModelicaSpecification#3473. Unfortunately, this generated a lot of controversy and at some point I moved on to other things and nobody seemed to cary any longer. Maybe we should refocus on it, once MSL 4.1.0 is released 😃 

In the meantime, we should get this fixed.

casella, 2025-03-03:
> The example has been changed from 4.0.0 to 4.1.0 ([01c3f90](https://github.com/modelica/ModelicaStandardLibrary/commit/01c3f90f1b0050045008122c15153008b029a03b)): 4.0.0 the machine is driven at constant speed, stop time is 0.04.
> 4.1.0 the machine is driven by a speed ramp, stop time is 0.32.

OK, so we should edit [creation.txt](https://github.com/modelica/MAP-LIB_ReferenceResults/blob/v4.1.0/Modelica/Electrical/Machines/Examples/SynchronousMachines/SMPM_NoLoad/creation.txt) accordingly, changing the two values of stopTime that show up in there, then have LTX re-generate the reference results and upload them to the 4.1.0 branch of the reference results repo.

> However, the diagram shown by [@casella](https://github.com/casella) is obtained from 4.1.0.

Sure. So, it should last 0.32 s, not 0.04 as currently shown.

> The difference stems from the equation in Modelica.Electrical.Machines.SpacePhasors.Blocks.ToPolar: y[2] = if noEvent(y[1] <= small) then 0 else Modelica.Math.atan2(u[2], u[1]); As far as I remember the "noEvent" is used to avoid a problem for detecting the angle of (x=0, y=0), i.e. for y[1]=0. IMHO the ramp from -pi to +pi should be a step.

It isn't, precisely because of the noEvent. We would see a step if there was a (recorded) event, so there would be two values a the same time point, one before the event and one after. But there are no events here, so we get the literal result. Which is not well defined whenever the angle is `pi + 2*k*pi`.

@AHaumer I think comparing angles when stuff rotates more than one turn is always going to give us problems, because of the discontinuous output. I would suggest to change the reference signals and use toSpacePhasor.y instead of toPolar.y. The information is basically the same, but the output of toSpacePhasor.y is continuous and always well-defined, so it won't give us problems.

[image]

Do you agree?

AHaumer, 2025-03-03:
@casella I agree: For now let us change the comparison signals, I'll prepare a PR.
However, it has to be possible to obtain proper angles even in case the shaft rotates more than one turn. 

In this case, the mechanical angle is not the problem, it is a problem of conversion from rectangular representation of a phasor to polar representation. Enclosed you find a small test example with an alternative formulation of ToPolar (you may switch between original implementation noEvent(... atan2) and an alternative without noEvent using acos (hopefully no division by zero).
The test setup is a phasor with constant angle but length following a ramp from -1 to 1.
Therefore just in the middle the angle has to jump by pi (not producing a ramp!).
We should keep an eye on that.
[Polar.zip](https://github.com/user-attachments/files/19060456/Polar.zip)

casella, 2025-03-05:
> [@casella](https://github.com/casella) I agree: For now let us change the comparison signals, I'll prepare a PR - see [#4560](https://github.com/modelica/ModelicaStandardLibrary/pull/4560)

OK.

> However, it has to be possible to obtain proper angles even in case the shaft rotates more than one turn.

To get a robust solution, we need to actually get a constantly increasing angle. No jumpts. 

Unfortunately, this requires some kind of "state" information, it can't be done by a function. I have some solutions that involve using a fast low-pass filter to keep that state information, we can discuss that in a separate issue if you want.

casella, 2025-03-05:
PR #4560 is merged in.

@Esther-Devakirubai please back-port to maint/4.1.x

@MatthiasBSchaefer, @GallLeo once that is done, please re-generate reference results.

casella, 2025-03-06:
> Agree. To obtain an "unwrapped" angle we need a PLL / ATO, see [#3914](https://github.com/modelica/ModelicaStandardLibrary/pull/3914)

Great! I missed that.

> no progress.

Let's restart the conversation on PR #3914 after releasing MSL 4.2.0. This is also interesting for me

casella, 2025-03-08:
#4565 back-ported #4560.

[14f18770](https://github.com/modelica/MAP-LIB_ReferenceResults/commit/14f18770c83756389864c5a75330a1fe779b8261)  updated the reference file.

closed 2025-03-08 by casella