modelica/ModelicaStandardLibrary #4382: Relevant enhancements to mention in the MSL 4.1.0 release notes

casella, 2024-04-09:
This [report](https://github.com/modelica/ModelicaStandardLibrary/pulls?q=is%3Apr+label%3Aenhancement+milestone%3AMSL4.1.0+is%3Aclosed) lists 51 PRs to MSL 4.1.0 marked with the "enhancement" tab. Many of them are rather trivial or technical, e.g. updating versions of library dependencies.

IMHO the ones worth mentioning in the Release Notes are: 
- improved 1D friction models (rotational #3662 and translational #3690)
- improved SignalPWM model #3679 
- added reading from CSV files in to ModelicaStandardTables #3691 
- added quasi static transformer model #3697 
- added 1D mechanical components with inverse torque-speed relationship #3702
- added ChopperBuckBoost power converter model #3774 
- added ideal rotational coupling #3786 
- added electrical impulse models #3831 
- added quasi-RMS sensors for single-phase systems #3876 
- improved PowerConverters.DCDC.HBridge model #3897 
- improved eddy current sources #3976
- improved FluxTubes #3993 
- new Exponentiation block #4006 
- new block SignalExtrema for continuous signals #4015 

@arunkumar-narasimhan, @Harisankar-Allimangalath, please take note
-

tobolar, 2024-04-10:
I can confirm for 
- [x] improved 1D friction models (rotational #3662 and translational #3690)
- [x] added 1D mechanical components with inverse torque-speed relationship #3702
- [x] improved eddy current sources #3976

In contrast,
- added ideal rotational coupling #3786 

is IMO not fully closed. It interrelates to #4019 which is still open!

casella, 2025-04-11:
I double-checked that all mentioned improvements are mentioned in the release notes. Also #3786 and #4019 are closed.

closed 2025-04-11 by casella